StefanSchippers/xschem #143: Changing loaded Libraries breaks alt+L label placement

Victor-Main, 2023-11-11:
I have added this two lines into my xschemrc in order to better control which libraries are loaded

```
set XSCHEM_LIBRARY_PATH :/usr/local/share/xschem/xschem_library/devices/
append XSCHEM_LIBRARY_PATH :/home/victor/.xschem/xschem_library/
```

However, Upon doing so and loading Xschem again attempting to create a label with alt+L it will not load the symbol. It attempts to load an ab_pin.sym symbol in the /home/victor forlder. However, if I allow the default libraries to be loaded  ALT+L will correctly load the lab_wire.sym present in the devices library.

[image]

I want to mention that the devices library is loaded and that I manually can insert a label symbol. However, using the hotkeys is much more efficient

StefanSchippers, 2023-11-11:
Thank you for reporting. The problem is caused by the trailing `/` in the set `XSCHEM_LIBRARY_PATH .../devices/` line. I will fix that.

StefanSchippers, 2023-11-11:
Ok this is fixed now. Thank you for reporting!

Victor-Main, 2023-11-11:
My pleasure. Thanks to you for the quick fix

closed 2023-11-11 by Victor-Main