StefanSchippers/xschem #148: PINORDER Symbol Attribute

kwmartin, 2023-12-06:
Currently, it seems that the pinorder is defined by going through the symbol pins (actually boxes on the pin level) pin by pin and using Shift-S to define a number. I am almost finished in converting my Viewdraw libraries to xschem (using a python program) and am trying to decide what to do with the PINORDER attribute in Viewdraw. Viewdraw uses this (when present) to order pins by names, assuming the pins have been named. Viewdraw also allows for *.pin files to be included in a directory to specify pinorder by names (one per line) as an alternative. Does xschem allow for a global  PINORDER attribute to be included in the symbols to specify the order using pin names without having to go through the symbol pins one by one, or is there an alternative method to specify pinorder using a single line?

StefanSchippers, 2023-12-06:
Xschem has a sim_pinnumber attribute trah determines the ordering of pins.
So the order of pins when netlisting a component is:
- the sorted order of pins if they have a sim_pinnumber attribute of if no such attribute exists on all pins:
- the order the pins are created and stored internally in xschem
As a last option you can specify individual pins in the symbol format string: if you have a NAND gate with inputs A, B, power pins  VCC, VSS, output Z you may specify:
`format="@name @@A @@B @@VCC @@VSS @@Z @symname"`
This pin ordering will be used both for the instance call:

`X1 A_NET B_NET VCC VSS Z_NET NAND2`
and in the .subckt definition:
```
.subckt NAND2 A B VCC VSS Z
...
...
.ends
```
see this manual page and search for sim_pinnumber:
https://xschem.sourceforge.io/stefan/xschem_man/symbol_property_syntax.html
and this tutorial:
https://xschem.sourceforge.io/stefan/xschem_man/tutorial_use_existing_subckt.html

The pinnumber attribute that may also be present on symbol pins defines the position of the pin in a package. This used for PCB design and does not have any effect on pin ordering in spice / vhdl / verilog netlists.

Stefan

kwmartin, 2023-12-19:
Thank you Stefan. I tried all the alternatives as you described and they all work. I also tried:
```
type=subcircuit
format="@spiceprefix@name @pinorder @symname"
verilog_format="@MODEL #(  ) @name ( @pinlist , @VDD , @VSS );"
template="name=x1   VDD=VDD VSS=VSS MODEL=DG_INV2 pinorder=\"@@OUT @@IN\""
generic_type=" VDD=string VSS=string MODEL=string pinorder=string"
verilog_extra="VDD VSS MODEL"
extra="VDD VSS pinorder"
verilog_netlist=true
spiceprefix=X
schematic=../../diglib28/sch/inv2_1x1.sch
verilog_primitive = true
```
which didn't work. I was hoping the @@variables would be substituted first as a workaround. This is not a big deal as I can control the order during the translation process.

On a separate issue: is it possible to add a system command when netlisting (I have a python script that fills in stress parameters based on reasonable guesses)? I'm guessing that defining a new Tcl command that calls exec might work? But how to have it executed just after netlisting and pass the the name of the netlist file to it might be nice?

Finally, in my translation from Viewdraw to xschem (mostly now working), I would like to have some of the fonts smaller and perhaps control on rotation and/or flip. For smaller fonts, I made a new pin-label symbol, but this doesn't seem to be be very clean; do you have something built in to control fontsizes and rotation/flipping? on pin and wire labels?
The more I use xschem, the better I like it; well done. P.S. I'm closing as you answered my pinorder question.

closed 2023-12-19 by kwmartin

StefanSchippers, 2023-12-19:
For the @@variables there is no recursive substitution. 

You can execute  an xschem command like:

` xschem netlist; execute_wait 0 gvim $netlist_dir/[file rootname [xschem get current_name]].spice
`
This example shows the produced netlist inside a gvim editor after netlisting.

`execute_wait` is an improved exec command. It will not block the xschem event loop (but stops the user from doing changes in the schematic until the child process is done)  so the schematic can be redrawn, moved / zoomed while the child process is still running.
There is also the `execute` procedure that does not block xschem at all. The '0' parameter tells xschem not to pop up a status dialog box when child process ends.

recent xschem has added an instance attribute text_size>n>=size,
example:
text_size_0=0.5

will set the symbol text with index 0 (position in the xschem data structure) 0 to size 0.5
For pins and labels the text of interest is always at position 0, so you can set placement of labels and in/out pins with custom text sizes.
[image]

You can anyway use exec as well. But xschem will freeze until the child finishes.