StefanSchippers/xschem #149: save result plot

fedeinthemix, 2023-12-09:
Is it possible to save an on schematic result plot as a PDF/EPS for use in documentation? I managed to export the whole schematic + plots, but, in my reports, I like to have the result plots separate from the schematic.

StefanSchippers, 2023-12-09:
you can hide the graphs (if no data is loaded) in Simulation->Graph menu. This way you can export the circuit only.

If you want a separate plot for the graphs you can copy/paste all graphs in a new empty schematic and after loading the raw file export the graphs to pdf. Another possibility is to zoom only the graphs and export to pdf.

You can also create a symbol with no I/O pins that has a corresponding  schematic containing only graphs. Descending into it will lead you to the simulation data. You can set attribute `spice_ignore=true` in the symbol so this symbol will be ignored for netlists. This allows you to do a hierachical pdf export with circuit and graphs in separate pdf pages.

fedeinthemix, 2023-12-09:
Thanks for the suggestions!
When I export to PDF the graphs do not show up. Is this a very recent feature? I'm using 3.4.4.
When exporting as SVG it works, but the resolution is quite low (see attachment). Is there any way to adjust it?
[image]

StefanSchippers, 2023-12-09:
for graphs in pdf export make sure you have libjpeg. If you run ./configure as explained in the [install](https://xschem.sourceforge.io/stefan/xschem_man/install_xschem.html) page
the following should appear:
```
=====================
Configuration summary
=====================

Compilation:
 CC:        gcc
 debug:     yes
 profiling: no

Paths:
 prefix:        /home/schippes
 user-conf-dir: ~/.xschem
 user-lib-path: ~/.xschem/xschem_library
 sys-lib-path:  /home/schippes/share/xschem/xschem_library/devices

Libs & features:
 tcl:       -ltcl8.6
 tk:        -ltcl8.6  -ltk8.6
 cairo:     yes
 libjpeg:   yes
 xcb:       yes

Configuration complete, ready to compile.
```
For svg and pdf graphs and embedded images are rendered as bitmaps. This because pdf and svg can not handle the amount of data of a vectorized graph, often having 100k or more points for every waveform.

I can add options to increase the size of the bitmap, though.
Will come back with some options for this.

StefanSchippers, 2023-12-09:
This for example seems a good definition (I have doubled the bitmap x/y sizes)
[image]

will add some options for quality.

fedeinthemix, 2023-12-09:
Thank you very much!

StefanSchippers, 2023-12-09:
latest xschem commit has a much better resolution of graphs in pdf and svg exports.
About your question for having graphs in a separate page I managed to add this symbol called for example graphs.sym:

[image]
the symbol has `type=primitive` and no format attribute:

[image]

This symbol will be traversed in a hierarchical pdf export but ignored for all netlist / simulation purposes.

See a sample pdf file produced.
[poweramp.pdf](https://github.com/StefanSchippers/xschem/files/13625873/poweramp.pdf)

fedeinthemix, 2023-12-10:
The resolution looks much better now, thanks.

Regarding exporting the graph only, the tricks that you explained are quite useful. But what I'd like, ideally, is the ability to specify a physical size for the figure, an not have any extra decoration such as the schematic name. Currently, it looks like the output is always A4 and the output always includes the schematic name at the top left.

StefanSchippers, 2023-12-11:
The schematic at the top left is now an option (default ernabled) 
If you specify:
`set ps_page_title 0`
in your xschemrc no schematic name will be printed on page.

Xschem has command line commands to print the area you want for svg and png:

xschem print svg plot.svg 0 0 1200 -1020 1880 -220
xschem print png plot.png 0 0 1200 -1020 1880 -220

the last 4 coordinates are the x1, y1, x2, y2 (upper-left point, lower-right point) if the area you want to print.
You can read the coordinates of the mouse pointer in the status bar.
the two zero coordinates are meaningful for png. If they are not zero you can specify a physical size in pixels  (w x h) of the generated png image.
If the w x h size has a different aspect ratio than the area you want to print xschem will arrange the printed area inside the w x h size.

The area to print is not available for ps/pdf  print:
xschem print pdf plot.pdf

the area shown in the window will be printed.

For embedding images in documents I think svg is the best choice.
I know it is possible to generate ps/pdf and not specify a paper size but only the bounding box of the objects as done with EPS.
Since postscript is a horrible language I will try to figure out if I remember how to do that.
If I manage to do that the pdf print command will also have a similar interface as svg:
xschem print pdf file.pdf 0 0 x1 y1 x2 y2

StefanSchippers, 2023-12-11:
I have verified that "bboxed" pdfs can be imported gracefully into libreoffice writer or libreoffice impress:
[image]

fedeinthemix, 2023-12-11:
Hi Stefan, that's great!

Just a comment: PostScript files meant to be embedded in other documents should be EPS (rather than PS which is meant to describe a whole document).  Therefore, when exporting a selection, it would probably make sense to export to EPS rather than PS.

If I try to use the generated PS (by simply setting the extension to .eps) in a LaTeX document
```LaTeX
\documentclass[11pt,a4paper]{article}
\usepackage{graphicx}

\begin{document}
\section{Intro}

This is a \ldots\\
\includegraphics[width=0.9\textwidth]{rc_lpf}

\end{document}
```
I get the following error
```bash
x/latex/l3backend/l3backend-dvips.def) (./test_doc.aux)

! LaTeX Error: Cannot determine size of graphic in rc_lpf.eps (no BoundingBox).

See the LaTeX manual or LaTeX Companion for explanation.
Type  H <return>  for immediate help.
 ...                                              
                                                  
l.8 \includegraphics[width=0.9\textwidth]{rc_lpf}
                                                 
? 
```
while, if I convert it to EPS with, e.g., ps2eps, it works fine.

StefanSchippers, 2023-12-11:
Yes I know. 
EPS is very complex to understand.
If I find some tutorial for humans may be I can add DSC comments or things like that to make eps consumers (like LaTeX) happy. 
If not I will proobably rely on ps2eps or some other tool to do the dirty job.

StefanSchippers, 2023-12-12:
I have added an EPS export function.
You need to select (by area, usually) the objects you want to export in the Encapsulated Postscript file, then:
File-> Image export ->EPS selection export.

Can you please try if LaTeX does read this EPS file? I have used the file produced by ps2eps as a reference and tried to do the same thing.

fedeinthemix, 2023-12-12:
It works fine. Thank you for your great support!

closed 2023-12-13 by fedeinthemix