StefanSchippers/xschem #150: clinking any buttons in prop window cause xschem cashing in v3.4.5

haoiyang, 2023-12-11:
Hi All:

I met an issue. When I tried to edit a symbol's property. I pressed "q" and edited in the prop windows.
Then, I clicked to save the modification, and xschem crashed.
I tried to click other buttons on the prop windows, and all buttons caused xschem to crash.
Could anyone help me to resolve this issue?

Thank you.

StefanSchippers, 2023-12-11:
Can you send the symbol you were trying to edit? I tried to do that and I can't reproduce the issue.

haoiyang, 2023-12-11:
Hi Stefan:

Thank you.
I opened a sch or a symbol view. Then I pressed "q" and clicked any button on it. Then, xschem crashed.
When I did "configure" the source code, It showed something not found.
#########
Found best fit 16 bytes wide float: long double
Checking for cairo... OK ('', '-I/usr/include/cairo -I/usr/include/glib-2.0 -I/usr/lib/aarch64-linux-gnu/glib-2.0/include -I/usr/include/pixman-1 -I/usr/include/uuid -I/usr/include/freetype2 -I/usr/include/libpng16' and '-lcairo')
Checking for cairo-xcb... OK ('#include <cairo-xcb.h>\n', '-I/usr/include/cairo -I/usr/include/glib-2.0 -I/usr/lib/aarch64-linux-gnu/glib-2.0/include -I/usr/include/pixman-1 -I/usr/include/uuid -I/usr/include/freetype2 -I/usr/include/libpng16' and '-lcairo')
Checking for xcb... not found
--- Detecting runtime (same as host)
#############
Besides, when I turned on xschem, the terminal showed some fails
##############
root@78337cc28325:/usr/src/xschem-3.4.5/src# xschem
tcleval(): evaluation of script: lindex $colors 0 failed
         : can't read "colors": no such variable
tcleval(): evaluation of script: lindex $colors 1 failed
         : can't read "colors": no such variable
tcleval(): evaluation of script: lindex $colors 2 failed
         : can't read "colors": no such variable
tcleval(): evaluation of script: lindex $colors 3 failed
         : can't read "colors": no such variable
##############
Are these possible sources to make xschem crash?
How should I do to fix it?

Thank you sincerely.

StefanSchippers, 2023-12-11:
Please send the complete scconfig/config.log file.
what system are you using?
ensure no xschem package is installed (sudo apt purge xschem)
in the xschem-3.4.5/src directory run .`/xschem` instead of `xschem`
Also go to this [manual](https://xschem.sourceforge.io/stefan/xschem_man/install_xschem.html) page and install all the required packages.

haoiyang, 2023-12-11:
Hi Stefan:

Thank you very much! 
I follow this "in the xschem-3.4.5/src directory run ./xschem instead of xschem"
Then, the issue is resolved, and no error shows anymore.

Thank you.

StefanSchippers, 2023-12-11:
If `xschem` fails and `./xschem` is OK it means you have an older xschem version installed on the system. Try `which xschem` to see where it is located.
You should remove the old installation, typically this is done with `sudo apt purge xschem`

StefanSchippers, 2023-12-11:
also running ./xschem inside the `src/` directory is usually done only for testing the program.
After compiling xschem you should do a `[sudo] make install` to install xschem on the system.

closed 2023-12-11 by StefanSchippers