StefanSchippers/xschem #151: A question I should know the answer to

kwmartin, 2023-12-23:
I should really be able to figure this out myself so please excuse my lack of skill, but:

I've been reading the section on Developer Info regarding Some Useful Script Examples:
"The following examples show the xschem commands one by one. In general you should create small TCL procedures to perform these tasks. This way you can optimize things, for example creating temporary variables holding the output of the various xschem ... commands."

Should these small TCL procedures be placed in the xschemrc file? Or somewhere else? I'm not sure what the recommended flow is when using custom commands? And should hot-keys be added to run them? Also, related, can the user add their own command to Simulation Configuration?

olisnr, 2024-01-09:
if You want to make TCL procedures that are project-specific, then a nice way is to using the `launcher.sym`. there are examples like the `rlc.sch` how to use it.

StefanSchippers, 2024-01-09:
In addition to placing stuff in schematics another possibility is to create a tcl file, say /path/to/file.tcl and add this filename in the tcl_files variable placed in xschemrc:
`set tcl_files /path/to/file.tcl `

the tcl_files can be a list of multiple files:
`set tcl_files {/path/to/file1.tcl  /path/to/file2.tcl} `

You can also directly add commands in xschemrc, using the postinit_commands variable; see the template xschemrc file in the src/ directory of xschem for comments.
the postinit_commands will be executed when xschem has completed initialization (while xschemrc instructions are sourced at beginning) so all commands, even the xschem specific ones are available.

You can add hotkeys easily:

```
bind . <KeyPress-6> {procedure}
bind . <KeyPress-Control-u> {puts Hello}
```
where `procedure` is some tcl proc defined as explained above.
Consider that numerous hotkeys are already taken by xschem so try not to shadow existing commands.

For adding simulator commands save the simulation configuration. A `simrc` file will be created in `~/.xschem`.
Edit that file, copy / paste the lines from an existing simulator, make the changes and save.
Here a 6th simulator is added (they are numbered 0...5) update the `sim(spice,n)` variable to 6 and add the `sim(spice,5,...)` as shown below.

```
#Specify the number of configured spice tools.
set sim(spice,n) 6 ;# number of configured spice tools
# Specify the default spice tool to use (first=0)
set sim(spice,default) 0 ;# default spice tool to launch
...
...
# specify tool command (cmd), name (name), if tool must run in foreground and if exit status must be reported
set sim(spice,5,cmd) {my_simulator "$N"}
set sim(spice,5,name) {My simulator}
set sim(spice,5,fg) 0
set sim(spice,5,st) 1
```

closed 2024-03-23 by StefanSchippers