StefanSchippers/xschem #153: Library Path Issue: It used to work

kwmartin, 2023-12-29:
Something has changed so library symbols are no longer found. See screen grab below.
I was running version 3.4.5. To be certain, I did a git pull which down loaded many changes, and then rebuilt. It is still at version 3.4.5, but now the most recent update; this did not cause anything to change.
Aside: the installation is still broken for user with a custom umask. I use umask 027, and after installing I still need to do:
cd /usr/local/share/xschem/
ssu
chmod 755 $(find . -type d)
chmod 644 $(find . -type f)
cd /usr/local/share/doc/xschem/
chmod 755 $(find . -type d)
chmod 644 $(find . -type f)
chmod a+x /usr/local/bin/xschem
I suggest something like this be added to the sudo make install?
Note in the screen grab below: The readonly entry in the "Path:" window is incorrect: it reads:
/home/martin/.xschem/proj_lib/devices and it should read /home/.xschem/xschem_library/devices
My is XSCHEM_LIBRARY_PATH is:
```
% puts $XSCHEM_LIBRARY_PATH                                   
/home/martin/.xschem/xschem_library/devices/:/home/martin/IC_Design/Viewdraw/cnvrtViewDrw/out/:/home/martin/IC_Design/Verilog/:/home/martin/.xschem/dig_lib/:/home/martin/.xschem/std_dig_lib/:
% 
```
This should work. I have tried many other changes to XSCHEM_LIBRARY_PATH and none I have tried work. Again, it used to work. If I prepend "/home/martin/.xschem/xschem_library/" to the Symbol entry widget, the symbol is found. Also, if I browse to the symbol, it is found.
My guess: the "Path" entry is set somewhere incorrectly, and saved between sessions. I can't find where; there's no mention of a "Path" variable in the xschemrc file. It could be due to the fact that I set the schematic attribute in my global symbol properties.
Finally, a suggestion:
For directories included in XSCHEM_LIBRARY_PATH, walk through the subdirectories and find all subdirectories that include any *.sch or *.sym files and include them in your library lookup table. My library structure, which I am stuck with as it needs to be compatible with my historic Viewdraw library structure, is set up with library_name/sym and library_name/sch subdirectories that include my sym and sch files, respectively. I get around xschem not finding my sch files by using attributes like: schematic=../../diglib28/sch/dr_3x2.sch (as you suggested) which are automatically included during conversion from Viewdraw, so this issue is not limiting for me, but I think my suggestion might make xschem's handling of library searches cleaner; just a suggestion.
Finally, finally, I have also a-periodically seen in the Symbol widget entry, sometimes the first character is missing, and sometimes "//" instead of just "/".
Finally, finally, finally, is the "Issues" tab the best place to make tentative suggestions or do you have a preferred location to make suggestions (such as on Sourceforge)?
Thank you.

[image]

StefanSchippers, 2023-12-29:
If you do a 'q' on a "missing symbol lab_pin" what is the "Symbol" reference?
if it is just `lab_pin.sym` then xschem should find it in `/home/martin/.xschem/xschem_library/devices`, the full path is obtained by adding `lab_pin.sym` to the first XSCHEM_LIBRARY_PATH compinent.
If the reference is "`devices/lab_pin.sym`" you **must** have a `/home/martin/.xschem/xschem_library` path element (without /devices).

If you have multiple directory levels you may add all the libraries, putting the deepest directories first so a symbol xxx.sym inside /a/b/c/d/ will be saved in the schematic as just `xxx.sym` if `/a/b/c/d` is listed in XSCHEM_LIBRARY_PATH and is listed before `/a/b/c` , otherwise the symbol reference will be saved as `d/xxx/sym`

For the umask problem you installed xschem using `sudo make install` so all files are owned by root and given your default umask you don't have access to these files. However if you install in your home directory you don't need to use `sudo`. All files will be owned by your unix account and there is no need to do a chmod.

StefanSchippers, 2023-12-29:
Another comment, if your schematics have absolute symbol references, if you set up correctly the XSCHEM_LIBRARY_PATH list of paths, load and force-save each schematic the absolute references will be removed.

StefanSchippers, 2023-12-29:
Another comment, if symbols and schematics are in dirrerent directories you may want to specify:
`set search_schematic 1`
in your xschemrc file. This way the schematic associated to a symbol will be searched in all paths listed in XSCHEM_LIBRARY_PATH.

StefanSchippers, 2023-12-29:
> For directories included in XSCHEM_LIBRARY_PATH, walk through the subdirectories and find all subdirectories that include any *.sch or *.sym files and include them in your library lookup table. 

This can not be done in general, if a user specifies [pwd] as one of the search paths xschem could walk the whole user home directory, adding directories that have nothing to do with the current project, and/or creating a very long list of directories. I think the automatic recursion is potentially dangerous. 
Even a complex project must have a countable number of directories such that they can be listed individually (you can get a list of dirs with `find . -type d`). Otherwise the project needs to be refactored somehow.

cnMuggle, 2024-01-07:
> If you do a 'q' on a "missing symbol lab_pin" what is the "Symbol" reference? if it is just `lab_pin.sym` then xschem should find it in `/home/martin/.xschem/xschem_library/devices`, the full path is obtained by adding `lab_pin.sym` to the XSCHEM_LIBRARY_PATH components in order until the file is found. If the reference is "`devices/lab_pin.sym`" you **must** have a `/home/martin/.xschem/xschem_library` path element (without /devices).

Thanks for the devices/ folder suggestions, it does work. I have a schematic created somewhere else, when I bring it into my current workspace, all the pin labels are shown as missing. It just shows /foss/design/ The default XSCHEM_LIBRARY_PATH only defines /foss/tools/xschem/64ca25b/share/xschem/xschem_library/
After I add /foss/tools/xschem/64ca25b/share/xschem/xschem_library/devices
The schematic symbol does recover.
[image]

StefanSchippers, 2024-01-07:
As a general rule put the deepest directories first
/foss/tools/xschem/64ca25b/share/xschem/xschem_library/devices
/foss/tools/xschem/64ca25b/share/xschem/xschem_library/

There is also a magic command to fix such issues.

if you have a missing `lab_pin.sym` placement and your path has `/foss/tools/xschem/64ca25b/share/xschem/xschem_library/`
it means the correct placement is `devices/lab_pin.sym`. You can issue the command `fix_symbols 1`.
It will add one directory component to unresolved symbols, transforming `lab_pin.sym` into `devices/lab_pin.sym`. This is done for all unresolved symbols, if they can be found 1 level beneath the XSCHEM_LIBRARY_PATH list of dirs.

closed 2024-01-26 by StefanSchippers