StefanSchippers/xschem #155: Bitblast Problem

kwmartin, 2024-01-02:
At work, I updated to the most recent changes and am now getting a bitblast error. I don't get this error at home (same version but not most recent changes), and all other files identical.
```
> xschem
% tclpropeval2: tcleval(`include "[abs_sym_path addr.v]")
---> path=->netlisting
tclpropeval2: tcleval(`include "[abs_sym_path regs.v]")
---> path=->netlisting
tclpropeval2: tcleval(`include "[abs_sym_path cmplt.v]")
---> path=->netlisting
tclpropeval2: tcleval(`include "[abs_sym_path rom_lu.v]")
---> path=->netlisting
tclpropeval2: tcleval(`include "[abs_sym_path mltply_25p125.v]")
---> path=->netlisting
tclpropeval2: tcleval(`include "[abs_sym_path mltply_4bits.v]")
---> path=->netlisting
netlist: source_file=/home/martin/.xschem/simulations/.QD_DDFS5_TB_423572, netlist_type=verilog
tcleval(): evaluation of script: netlist {/home/martin/.xschem/simulations/.QD_DDFS5_TB_423572} noshow {QD_DDFS5_TB.v} failed
         : can't read "verilog_bitblast": no such variable
tclpropeval2: tcleval(`include "[abs_sym_path addr.v]")
---> path=->netlisting
tclpropeval2: tcleval(`include "[abs_sym_path regs.v]")
---> path=->netlisting
tclpropeval2: tcleval(`include "[abs_sym_path cmplt.v]")
---> path=->netlisting
tclpropeval2: tcleval(`include "[abs_sym_path rom_lu.v]")
---> path=->netlisting
tclpropeval2: tcleval(`include "[abs_sym_path mltply_25p125.v]")
---> path=->netlisting
tclpropeval2: tcleval(`include "[abs_sym_path mltply_4bits.v]")
---> path=->netlisting
netlist: source_file=/home/martin/.xschem/simulations/.QD_DDFS5_TB_423572, netlist_type=verilog
tcleval(): evaluation of script: netlist {/home/martin/.xschem/simulations/.QD_DDFS5_TB_423572} show {QD_DDFS5_TB.v} failed
         : can't read "verilog_bitblast": no such variable
```

StefanSchippers, 2024-01-02:
Thank you, 
this was due to a typo (extra character deletion) in a global variable in a commit done on 28 Dec 2023
It is fixed now.
Thank you for pointing this out
Happy 2024!
[image]

kwmartin, 2024-01-03:
Thanks Stefan, I confirm I'm now back up and running with the latest git pull

closed 2024-01-03 by kwmartin