StefanSchippers/xschem #157: copied content cannot be overwrite by new copy via mouse right click, not consistent behavior

cnMuggle, 2024-01-07:
I am using 3.4.5 now from iis-osic. I noticed the copy/paste behavior in xschem is not function correctly and not consistently

I am not able to select the text to copy in the "Tools->Execute TCL command"
[image]

here is what I am trying to do, to copy this text from this TCL command input and paste it to xschemrc. I cannot copy it either by right click or ctrl+c. 
[image]

If I press mouse middle button. I think depending on when my mouse pointer is located at. something from originally selected is first copy and then paste into the existing text. 
[image]

Here is what I get from the Windows paste board:
**_PATH "$XSCHEM_LIBRARY_PATH:/foss/tools/xschem/64ca25b/share/xschem/xschem_library/devices/"**

here is what I get from the Linux paste board. It's something I previously copied from xschem text window.
[image]

I guess maybe there is no copy/paste for TCL expression input?

StefanSchippers, 2024-01-07:
Select all the text in the Tools -> execute TCL command dialog box (this is done by going to the first character, press the left mouse button, drag the mouse to the end of the text, releasing the left button and press 'ctrl-c'.

Then open the xschemrc in a text editor and do 'paste' or press middle mouse button or press 'ctrl-v'
Do not close xschem before completing the paste operation because the selection is handled by the application by X11 standards.

cnMuggle, 2024-01-07:
> Select all the text in the Tools -> execute TCL command dialog box (this is done by going to the first character, press the left mouse button, drag the mouse to the end of the text, releasing the left button and press 'ctrl-c'.
> 
> Then open the xschemrc in a text editor and do 'paste' or press middle mouse button or press 'ctrl-v', depending on the editor. Do not close xschem before completing the paste operation because the selection is handled by the application by X11 standards.

Thanks, Stefan. If you first copy half of the text and paste them somewhere else, then come back to copy the text all and then paste somethwere, I will only see the half text copied. Somehow the behavior of copy/paste is not consistent. 

In this example below, I first copy $XSCHEM_LIBRARY_PATH and then I want to copy "puts $XSCHEM_LIBRARY_PATH"
but I only get $XSCHEM_LIBRARY_PATH pasted.
[image]

StefanSchippers, 2024-01-07:
I think you are using some virtual machine, may be there are some settings to handle sharing selection / clipboard with the host / guest Operating System. I don't see these issues.

cnMuggle, 2024-01-07:
> I think you are using some virtual machine, may be there are some settings to handle sharing selection / clipboard with the host / guest Operating System. I don't see these issues.

Just confirmed as well. It seems the Windows docker container issue of iss-osic which having xschem.

cnMuggle, 2024-01-07:
closed as issue related with OS or container

closed 2024-01-07 by cnMuggle