StefanSchippers/xschem #159: Ports Not Netlisted

yrrapt, 2024-01-29:
I have an issue where two particular ports (`chop_p_i` and `chop_n_i`) are not being netlisted into the subckt definition.

The schematic file is:

```
v {xschem version=3.4.5 file_version=1.2
}
G {}
K {}
V {}
S {}
E {}
C {devices/ipin.sym} 0 -60 0 0 {name=p4 lab=chop_p_i}
C {devices/ipin.sym} 0 -40 0 0 {name=p5 lab=chop_n_i}
C {devices/ipin.sym} 0 0 0 0 {name=p2 lab=gnd_i}
C {devices/ipin.sym} 0 -400 0 0 {name=p1 lab=vdd_1v8_i}
C {devices/ipin.sym} 0 -320 0 0 {name=p6 lab=ibias_10u_p_i[2:0]}
C {devices/ipin.sym} 0 -230 0 0 {name=p3 lab=in_p_i}
C {devices/ipin.sym} 0 -210 0 0 {name=p7 lab=in_n_i}
C {devices/opin.sym} 1850 -200 0 0 {name=p8 lab=out_n_o}
C {devices/opin.sym} 1850 -240 0 0 {name=p9 lab=out_p_o}
```

The ends up with a netlist (with LVS mode enabled):

```spice
.subckt test_circuit vdd_1v8_i out_p_o out_n_o ibias_10u_p_i<2>,ibias_10u_p_i<1>,ibias_10u_p_i<0> gnd_i in_p_i in_n_i
*.ipin chop_p_i
*.ipin chop_n_i
*.ipin gnd_i
*.ipin vdd_1v8_i
*.ipin ibias_10u_p_i<2>,ibias_10u_p_i<1>,ibias_10u_p_i<0>
*.ipin in_p_i
*.ipin in_n_i
*.opin out_n_o
*.opin out_p_o
**.ends
.end
```

I have tried to debug the cause but have not yet been able to determine what is causing the issue.

StefanSchippers, 2024-01-29:
Hi, Thomas, try to see if you have more than one test_circuit.sch in your search paths.
Are you netlisting a testbench that instantiates test_circuit.sym?
or are you netlisting test_circuit.sch directly?
in the first case edit the symbol attributes and look at the format attrinute. This tells xschem the port list.

yrrapt, 2024-01-30:
Hi Stefan,
I found the problem, the symbol did not have those pins.
Adding the pins to the symbol fixed the problem.
So this was user error.
Thanks,
Thomas

closed 2024-01-30 by yrrapt

StefanSchippers, 2024-01-30:
@yrrapt  If you run a netlist from a script, like:

```
$ xschem  -rnqx tb_opamp_stef.sch >/dev/null
-----------/home/schippes/.xschem/xschem_library/opamp_stef/tb_opamp_stef.sch
Warning: open net: VCC
-----------/home/schippes/.xschem/xschem_library/opamp_stef/opamp_stef.sch
-----------/home/schippes/.xschem/xschem_library/opamp_stef/tb_opamp_stef.sch
Error: Symbol opamp_stef.sym: schematic pin: VCC not in symbol
Error: Symbol opamp_stef.sym has 4 pins, its schematic has 5 pins
```
you get on stderr messages telling if something is wrong regarding to sch/sym port matching.
option `-r` is important since it prevents xschem from closing stdin/stdout/stderr (this is necessary if  multiple xschem instances are launched from the same controlling terminal).