StefanSchippers/xschem #160: Help, someone. stange color of  sky130 tt models block.

timibacco, 2024-02-03:
Hi , can anyone help explain why the tt models block for the sky130 tops is **red** on xschem startup. Red is not a good color. Can someone help out?
This is a screenshot though.
[image]
### 

Plus I also get this warning as well.At Xschem's startup as well.
```
Warning: PDK_ROOT env. var. not found or empty, trying to find an open_pdks install
open_pdks installation: using /usr/local/share/pdk
SKYWATER_MODELS: /usr/local/share/pdk/sky130A/libs.tech/ngspice
SKYWATER_STDCELLS: /usr/local/share/pdk/sky130A/libs.ref/sky130_fd_sc_hd/spice
WARNING: wrong line layer
WARNING: wrong line layer
WARNING: wrong line layer
WARNING: wrong line layer
WARNING: wrong line layer
WARNING: wrong line layer
WARNING: wrong line layer
WARNING: wrong line layer
WARNING: wrong line layer
WARNING: wrong line layer
WARNING: wrong line layer
WARNING: wrong line layer
WARNING: wrong line layer
WARNING: wrong line layer
WARNING: wrong line layer
WARNING: wrong line layer
WARNING: wrong line layer
```

Seems like an ngspice configuration, I guess, but I really don't know what resources to check out anymore.

StefanSchippers, 2024-02-03:
You have an old xschem version installed on the system.
Remove the old xschem:
sudo apt purge xschem
then install xschem sources from git and build the program. Read the instructions here:
https://xschem.sourceforge.io/stefan/xschem_man/install_xschem.html

timibacco, 2024-02-04:
Thanks very much. Worked like Xschem.

closed 2024-02-04 by timibacco