StefanSchippers/xschem #161: Pin Types for Auto Generated Supplies

kwmartin, 2024-02-05:
I have the following global attributes for a symbol:
```
type=subcircuit
format="@spiceprefix@name @pinlist VDD VSS @symname"
extra="VDD VSS"
verilog_format="mltply #( @nbits) ) @name ( @pinlist , @VDD , @VSS );"
verilog_extra="VDD VSS"
verilog_netlist=true
template=" nbits=18 VDD=VDD VSS=VSS"
generic_type=" nbits=int"
spiceprefix=x_?
schematic=../../dig_lib/sym/mltply.sch
```
When I generate a verilog netlist the added pins, VDD and VSS are given inout direction. This causes an iverilog error of:
... error: VDD Unable to assign to unresolved wires.
When I edit the netlist and change the directions to input, these errors go away. In addition, the "parameter" is not added to the module. Also, something in my global variables is causing lines near the netlist end just containing "?" which gives iverilog syntax errors unless I delete the lines. This doesn't happen on other testbenches, so I'm guessing I have something wrong in my global attributes. Some of the netlist lines are shown below. Any help is appreciated. Thank you.

```
// expanding   symbol:  sym/mltply.sym # of pins=5
// sym_path: /home/martin/.xschem/dig_lib//sym/mltply.sym
// sch_path: /home/martin/.xschem/xschem_library/devices//../../dig_lib/sym/mltply.sch
`timescale 1ps / 1fs
module mltply
(
  input wire [15:0] In,
  input wire [17:0] K,
  input wire Init,
  input wire Clk,
  output wire [17:0] Out,
  inout wire VDD,
  inout wire VSS
);
...
```

StefanSchippers, 2024-02-05:
For subcircuits you don't need a verilog_format attribute. xschem will correctly generate the subcircuit instance lines.
This is indeed different from spice netlist where the format attribute is always necessary. 
Since VHDL and Verilog instance calls are more verbose xschem does this on its own.
If type=subcircuit do not use verilog_format (just rename to xverilog_format, so it will be there if you ever need it)

Currently there is no way to specify port direction of pins specified via attributes. They default to 'inout'. I will add a direction specifier.

If you get '?' in a netlist it means some tcl evaluation (attributes in tcleval(....) ) failed, for example if you are using $xxx and global tcl var xxx is not defined. 

Anyway if you can provide a simple test case (provide also the symbols and subschematics if any) I will investigate more.

Thanks for now.

StefanSchippers, 2024-02-06:
It is now possible to specify directions for port specified as attributes using the `verilog_extra_dir attribute`.
```
type=subcircuit
format="@spiceprefix@name @pinlist VDD VSS @symname"
extra="VDD VSS"
verilog_format="mltply #( @nbits) ) @name ( @pinlist , @VDD , @VSS );"
verilog_extra="VDD VSS"
verilog_extra_dir="VDD=input VSS=input"
verilog_netlist=true
template=" nbits=18 VDD=VDD VSS=VSS"
generic_type=" nbits=int"
spiceprefix=x_?
schematic=../../dig_lib/sym/mltply.sch
```

closed 2024-03-23 by StefanSchippers