StefanSchippers/xschem #164: problem with PDF export:

olisnr, 2024-02-16:
i have a problem with PDF export. Xschems says:

```
ich@x13g3:~$ xschem
xschem [~] Raw file data read: /ALL/Xschem/design/simulation/jfet_input_new_buffer_20_alternativer_buffer__gut.raw
points=1, vars=73, datasets=1 sim_type=op
Raw file data read: /ALL/Xschem/design/simulation/jfet_input_new_buffer_20_alternativer_buffer__gut.raw
points=6416, vars=74, datasets=2 sim_type=dc
Raw file data read: /ALL/Xschem/design/simulation/jfet_input_new_buffer_20_alternativer_buffer__gut.raw
points=209693, vars=74, datasets=1 sim_type=tran
Raw file data read: /ALL/Xschem/design/simulation/jfet_input_new_buffer_20_alternativer_buffer__gut.raw
points=1821, vars=148, datasets=3 sim_type=ac
free_rawfile(): clearing data
free_rawfile(): clearing data
free_rawfile(): clearing data
free_rawfile(): clearing data
Raw file data read: /ALL/Xschem/design/simulation/jfet_input_new_buffer_20_alternativer_buffer__gut.raw
points=1, vars=73, datasets=1 sim_type=op
Raw file data read: /ALL/Xschem/design/simulation/jfet_input_new_buffer_20_alternativer_buffer__gut.raw
points=6416, vars=74, datasets=2 sim_type=dc
Raw file data read: /ALL/Xschem/design/simulation/jfet_input_new_buffer_20_alternativer_buffer__gut.raw
points=209693, vars=74, datasets=1 sim_type=tran
Raw file data read: /ALL/Xschem/design/simulation/jfet_input_new_buffer_20_alternativer_buffer__gut.raw
points=1821, vars=148, datasets=3 sim_type=ac
done
problems converting postscript to pdf: **** Warning: glyf overlaps cmap, truncating.
Use 'exit' to close the program
```
[jfet_input_new_buffer_20_alternativer_buffer__gut.sch.tar.gz](https://github.com/StefanSchippers/xschem/files/14309861/jfet_input_new_buffer_20_alternativer_buffer__gut.sch.tar.gz)

StefanSchippers, 2024-02-16:
I have generated pdf export and svg export. These are the files. Can you check if they load ok? I did not get any warning. 

NOTE: I have replaced olipwr.sym and olignd.sym with vdd.sym and gnd.sym, because these two models were missing. If you can please submit these two symbols so I can try again.
[jfet_input_new_buffer_20_alternativer_buffer__gut.pdf](https://github.com/StefanSchippers/xschem/files/14310234/jfet_input_new_buffer_20_alternativer_buffer__gut.pdf)

[image]

For the svg images I know that github is broken and does not render correctly. Just open with the gnome image viewer (eog) or with the browser directly (firefox /path/to/svg_image.svg)

StefanSchippers, 2024-02-16:
Also can you try to generate a pdf / svg from a very simple circuit?
still problems?

olisnr, 2024-02-16:
also: 
.SVG works on simpler circuits. may be the wires are getting to thin in bigger drawings, and are falling out of the files :)
.EPS woks on both files
.PDF dont write anything on my current computer. 

i think it could be some specific problem of my installation. i did compile the newest Xschem from github. may be i should try to reinstall some PDF/PS libraries?

olisnr, 2024-02-16:
to the .SVG
if i open with inskscape, it seems that the wires are simple to small to be rendered. may be You can export it different?

[image]

StefanSchippers, 2024-02-16:
For pdf you need ghostscript which provides the ps2pdf command. Xschem relies on a conversion program (defined in xschemrc to_pdf variable, default is ps2pdf) .If you don't have it you can try to export plain postscript. 
Just change the extension of the destination file to .ps instead of .pdf. You can then see if the ps is valid and generated.

olisnr, 2024-02-16:
.PS works

StefanSchippers, 2024-02-16:
Yes the svg problem is that lines are too thin: open the SVG file with an editor and change these lines:
stroke-width: 0.00844839;
to:
stroke-width: 0.844839;

Do wires appear? 

What is the resolution / screen size of the xschem window when exporting?

StefanSchippers, 2024-02-16:
I am trying various setups (window sizes / zoom) but i can not get a svg file with lines that thin. Are you generating the export from a script (using --svg, --plotfile , ....)?

olisnr, 2024-02-16:
no, i simply used the menue `File :: image export :: SVG export`.

if i use ps2pdf from command-line, its working, even if it prints the same warning like Xschem:

```
ich@x13g3:~$ ps2pdf '/home/ich/jfet_input_new_buffer_20_alternativer_buffer__gut.ps' 
**** Warning: glyf overlaps cmap, truncating.
```

olisnr, 2024-02-16:
changeing the xshemrc dont fix the .PDF export

StefanSchippers, 2024-02-16:
For the pdf ensure `gsfonts-x11` package is installed. I can not do much more on this , since it is a ghostscript issue. 
Usually this happens when custom fonts / non-ascii accented characters are used. as i see here:
https://www.experts-exchange.com/questions/28159547/Converting-postscript-to-pdf-font-problem.html

StefanSchippers, 2024-02-16:
I have done  commit to avoid having too small line widths in svg exports. Please check.

olisnr, 2024-02-16:
.SVG is perfekt. .PDF still not working, but on an other it works. do You have an idea how i could debug it?

StefanSchippers, 2024-02-16:
> but on an other it works. 
1. Do you mean on another system or another schematic?
2. If you export the echematic that does not work to PS instead of PDF and then manually convert to pdf with ps2pdf does it work?

olisnr, 2024-02-16:
one one computer i can export .PDF on the other not. both have the same ubuntu 22.04 based OS. i tried to reinstall gostscript ant the x11-fonts. the command-line-tool works on both systems, it seems to me, that on one system the ps2pdf command isnt correct called from Xschem. i would like to find out why. 

is it possible to run the xschem.tcl in a debuger?

StefanSchippers, 2024-02-16:
Try to see of on the failing system there is another ps2pdf around:

sudo find / -type f -name ps2pdf -ls

compare the file size on both systems.

StefanSchippers, 2024-02-16:
on my system /usr/bin/ps2pdf is the following script:
```
version=14
ps2pdf="`dirname \"$0\"`/ps2pdf$version"
if test ! -x "$ps2pdf"; then
        ps2pdf="ps2pdf$version"
fi
exec "$ps2pdf" "$@"
```
this will call /usr/bin/ps2pdf14
that is the following script:
```
ps2pdfwr="`dirname \"$0\"`/ps2pdfwr"
if test ! -x "$ps2pdfwr"; then
        ps2pdfwr="ps2pdfwr"
fi
exec "$ps2pdfwr" -dCompatibilityLevel=1.4 "$@"
```
this will invoke `/usr/bin/ps2pdfwr dCompatibilityLevel=1.4`

The /usr/bin/ps2pdfwr is yet another script that calls ghostscript (/usr/bin/gs) witha bunch of options.

StefanSchippers, 2024-02-16:
I think I have found the reason for the failure.
If the ps2pdf command spits out some **warning** on standard error xschem interprets this as a **failure**. Will fix that.

StefanSchippers, 2024-02-16:
Ok, last commit hopefully fixes the issue. Even if ps2pdf prints some warnings on stderr it will not be interpreted as a failure.
If you have a chance please test.
Thank you for the report and feedback

olisnr, 2024-02-27:
newest version fixed it. thanks a lot!

closed 2024-02-27 by olisnr