StefanSchippers/xschem #166: [RFC] live op analysis + annotation

qpwo, 2024-02-16:
Would be nice if there was a feature to rerun op analysis and re-annotate after every change you make to the circuit. Could speed up iteration a decent amount.

StefanSchippers, 2024-02-17:
This is dangerous. On a big circuit an OP analysis takes time, it is not uncommon for a user to be faster changing a circuit than the simulator to complete. This must be handled in some way, for example stalling the editor until the OP has completed or killing the running sim if a change is made while previous OP still running. All these cases are difficult to make robust due to race conditions. 

You can add a `launcher.sym` and bind the following command to it:

`tclcommand="xschem netlist; xschem simulate; xschem annotate_op"`

For the above to work you must configure simulation to be launched in foreground mode, otherwise `xschem simulate` will not block waiting for simulation to finish. Go to `simulation -> Configure simulators and tools` and set the 'fg' (Foreground) flag for the selected simulator.

If you are running interactive ngspice sims remember to put a `quit 0` instruction at the end of your commands to quit the simulator, otherwise ngspice will keep running waiting for user input.

StefanSchippers, 2024-02-17:
Another smarter way is to set a callback to the simulate command:
`tclcommand="xschem netlist; simulate {xschem annotate_op}"`
In this case the annotation is given as callback argument to the simulate procedure. this callback will be scheduler after completion of the simulation, all in the background.

qpwo, 2024-02-18:
What syntax is this callback from?

StefanSchippers, 2024-02-18:
It is just an argument to the simulate procedure. In this case the argument is used as a command to run when simulation is completed/

qpwo, 2024-02-18:
Ah ok that makes more sense

qpwo, 2024-02-18:
Ah so I couldn't get it to work but it's my fault for using `gnome-terminal`. For example `gnome-terminal -- sleep 1; echo done` will echo done immediately. Will work it out on my own. Thank you for your help.

closed 2024-02-18 by qpwo