StefanSchippers/xschem #167: [RFC] click on symbol parameter to change it

qpwo, 2024-02-16:
For example, with this voltage source, it would be nice if I could just click the `1` and type `2` then click away or hit enter.

[image]

Would pair with #166 .

Speculation on details of how this could be implementable:

* Entire value is replaced with what you type; no fancy text cursor or anything
* Maybe a symbol's text field is only click-instant-editable if it has exactly one `@variable`. (Otherwise, xschem wouldn't know what you want to change.)
* Maybe disable by default
* Maybe regular symbol hitbox would be a bounding rectangle around all non-text graphics. Each text field's hitbox would be the part of its bounding rect outside of the main symbol hitbox.
* Could scroll on value to change it up and down super quickly.

StefanSchippers, 2024-02-16:
> For example, with this voltage source, it would be nice if I could just click the `1` and type `2` then click away or hit enter.

Currently double clicking on an instance brings up the edit attribute dialog box.
Double clicking on a value selects it. Typing something replaces the value.
Clicking in the schematic empty space applies the change closing the dialog.
Clicking another element will apply the change on previous element and update the dialog box with the new element attributes.

Before doing other changes I think a nice thing to add is to have the `value` attribute selected and the textbox focused after double clicking. This saves a (precision) double click to select text to replace.

[image]

qpwo, 2024-02-16:
Yes putting the cursor selection there sounds like 90% of the value for 10% of the effort!

qpwo, 2024-02-16:
Oh if #166 is feasible: Another checkbox in the Edit Properties window ("apply changes while typing") could maybe add another 2x speedup

StefanSchippers, 2024-02-17:
I have done the change:
When double clicking an element, if it has value= or lab= fields then select the text after the '=' and position insertilon cursor. 
To change a vsource value with minimal GUI actions:

- Double click the component (or select, press 'q')
- Type new value
- Click outside in an empty area of the schematic or click another instance to edit or type **Shift+Enter**

qpwo, 2024-02-18:
Awesome!!

Tiny pain point: if I move my mouse 1 pixel in the main xschem area, then the popup window loses focus.

https://github.com/StefanSchippers/xschem/assets/10591373/3a816a1b-4a45-400d-8710-17b5c183b725

Is there any easy way to increase that tolerance or a setting to prevent mouse-move from changing focus?

(My general window manager does not change focus on mouse move, if that's relevant.)

No worries if it's too annoying to change!

StefanSchippers, 2024-02-18:
I will try to set my window manager to 'click to focus' and see if I can avoid that losing of focus. Thank you

StefanSchippers, 2024-02-18:
@qpwo Is your xschem behaving differently than in this video?

https://github.com/StefanSchippers/xschem/assets/69359491/0c4cc6c3-782b-4998-8517-5f0ddf092d3b

qpwo, 2024-02-20:
I just pulled and rebuilt. The main xschem window is still grabbing focus on mouse move on my machine. Perhaps this is beyond xschem's control? Or I'm missing some setting? (Haven't noticed mouse-move-focus in any other apps on my system though.)

StefanSchippers, 2024-02-20:
You are right.

There is an option in xschemrc:
`set autofocus_mainwindow 0`
to disable this main window focus grabbing.
By default this setting is enabled.

Sorry I forgot this long standing option. (I also use this disabled autofocus setting).

Can you lease add/change this option in your xschemrc and test?

qpwo, 2024-02-20:
Working perfectly! Super rapid changes!

closed 2024-02-20 by qpwo