StefanSchippers/xschem #168: [RFC] Make filename (without extension) available as environment variable to ngspice

qpwo, 2024-02-16:
Problem: I copy or rename circuit1.sch to circuit2.sch and make some changes and do my sim but the plots aren't working because I was doing `write circuit1.raw`!

Solution: Xschem could set `$base` and your ngspice code can say `write {$base}.raw`. Then schematics are fully copyable/renamable!

I have my simulator in xschem configured to

```sh
gnome-terminal -- bash -c "base=`basename $N .spice` ngspice -i $N -a"
```
But if `$base` was enabled by default then the skywater example schematics could use it and it would be a lot easier for newbs to copy them and make changes.

StefanSchippers, 2024-02-16:
This is available as a simple tcl command: 
in the xschem terminal: 
`xschem get current_name` returns the schematic name, like `solar_panel.sch`
`file rootname [xschem get current_name]` returns `solar_panel`

So in your code block you can use this command to write the raw file with the same name as the schematic:

[image]

Note the whole command is wrapped inside a `tcleval(....)` construct. This will invoke a round of tcl substitution to resolve `$tcl_variables` and `[tcl commands]`
I will also add a `sch_basename` global TCL variable so the expression `[file rootname [xschem get current_name]].raw`
can be simplified as `$sch_basename`. Escaping can be used if you want a literal $something that must not be substituted, like `\\$something.raw`

 _(for the very nerd: double escaping is needed because the whole string is included in double quotes and \" can be used to enter a literal quote, so a `\\` will enter a literal `\` that will escape the $ to the TCL interpreter. If you don't get this quirk no problem, just ignore this detail :-) )_

qpwo, 2024-02-18:
Great seems to work:

[image]

closed 2024-02-18 by qpwo

qpwo, 2024-02-18:
Oh great I don't even need the `[return ...]`.

[image]

StefanSchippers, 2024-02-18:
Correct, the whole expression inside tcleval( ..... ) is passed to a tcl `subst` command. The subst command takes the input string and returns the string with [ ....] the return value of the associated tcl command and $xxx as the value for xxx variable.
so` tcleval(test [return tcl] $x) `will return `test tcl substitution`, assuming `x` variable is set to `substitution`