StefanSchippers/xschem #169: [RFC] ctrl-click spice code to run it

qpwo, 2024-02-16:
Problem: I want to quickly try different sims. I don't think the editor has a shortcut for commenting(?). No keyboard shortcut for toggle-spice-ignore either I think.

Solution: ctrl-click code symbol to generate netlist and execute it.

Editor shortcut for commenting or keyboard shortcut for toggle-spice-ignore could also solve this.

qpwo, 2024-02-16:
(Ok that's my last RFC for now!)

StefanSchippers, 2024-02-17:
There is such command, it is **Properties -> Toggle *_ignore attribute on selected instances**.

You can select multiple instances and after doing the menu command all selected elements will be toggled (ignored / not ignored)
Will look for a free bind key to associate (there is currently no keybind for that).

StefanSchippers, 2024-02-17:
You can ctrl-click the code symbol to trigger netlist and simulation:
Add this:
`tclcommand="xschem netlist; simulate"`

StefanSchippers, 2024-02-17:
So if I understood your reauest the sequence is:

- enable the code block (and disable others if any) by selecting and pressing `Shift-T`
- ctrl-click the (now enabled) code block to run the tcldommand attribute as explained above.

If a command block is disabled (spice_ignore=1) you can enable it before generatng a netlist:
`xschem setprop instance COMMANDS spice_ignore 0`
you can then create a netlist:
`xschem netlist`
and finally disable the code block again:
`xschem setprop instance COMMANDS spice_ignore 1`

StefanSchippers, 2024-02-17:
[image]

qpwo, 2024-02-18:
Shift-T is perfect! Much appreciated. That tcl recipe is handy too.

qpwo, 2024-02-18:
TIL that you can put `tclcommand=` on a `code_shown.sym`

closed 2024-02-18 by qpwo

StefanSchippers, 2024-02-18:
You can put that on any symbol to execute the relevant command.
You can also set a `url=....` attribute to open a web page or local document when ctrl-clicking the instance. Xschem will figure out what application to call (browser for http addresses, pdf viewer for pdf files. In cases you need to tell xschem what to use to open the provided url you can add a `program=...` to specify the application.

Do not set both tclcommand and url/program on the same symbol