StefanSchippers/xschem #170: pip install or conda install -c conda-forge xschem

joamatab, 2024-02-17:
As we have enabled [Schematic driven layout](https://gdsfactory.github.io/gplugins/notebooks/20_schematic_driven_layout.html) in [gdsfactory](https://gdsfactory.github.io/gdsfactory/) it would be great to have an extension for xscheme, so that we can automatically create symbols from our PDK Pcells

How could we pip install or conda install -c conda-forge xschem ?

@proppy

joamatab, 2024-03-02:
I found it [here](https://anaconda.org/litex-hub/xschem)

```
conda install litex-hub::xschem

```
but seems to have some issues to install

[image]

qpwo, 2024-05-25:
seems to be working now

StefanSchippers, 2024-09-05:
Don't know if I have to do something here... closing for now.

closed 2024-09-05 by StefanSchippers