StefanSchippers/xschem #171: support RF simulators

joamatab, 2024-02-17:
How could we enable [sax](https://flaport.github.io/sax/) and [scikit-rf](https://github.com/scikit-rf/scikit-rf/issues/352) for RF circuit simulations?

@flaport

StefanSchippers, 2024-02-17:
Are these RF packages using s SPICE simulator or do they provide their own simulation engine?

joamatab, 2024-02-18:
I mostly use sax and we could easily write a command line script  to simulate the netlist, how could we call that and retrieve the results?

StefanSchippers, 2024-02-18:
So what you want to do is to have some Opto/RF symbols in xschem, create a circuit (by placing and connecting these elements) and get the circuit description that SAX can read and simulate?
Should the definition of symbols be in some external file? or embedded in some way in the xschem symbols?

It is possible to generate only the structural netlist (how things are connected in the circuit), like in: 
```
mzi, _ = sax.circuit(
    netlist={
        "instances": {
            "lft": coupler,
            "top": waveguide,
            "rgt": coupler,
        },
        "connections": {
            "lft,out0": "rgt,in0",
            "lft,out1": "top,in0",
            "top,out0": "rgt,in1",
        },
        "ports": {
            "in0": "lft,in0",
            "in1": "lft,in1",
            "out0": "rgt,out0",
            "out1": "rgt,out1",
        },
    }
)
```
or provide also the definitions, like:
```
...
...
 def coupler(coupling=0.5):
    kappa = coupling**0.5
    tau = (1-coupling)**0.5
    sdict = sax.reciprocal({
        ("in0", "out0"): tau,
        ("in0", "out1"): 1j*kappa,
        ("in1", "out0"): 1j*kappa,
        ("in1", "out1"): tau,
    })
    return sdict
...
...
```
In the mean time I will learn a bit about Sax.

joamatab, 2024-02-19:
we use a YAML syntax to describe the layout (netlist + placement), we also can use YAML as a schematic representation in our [Schematic driven layout](https://gdsfactory.github.io/gplugins/notebooks/20_schematic_driven_layout.html)

ideally from xschem we can extract the SPICE netlist, then convert it to YAML format which can be simulated by [SAX](https://gdsfactory.github.io/gplugins/notebooks/sax_01_sax.html), then use gdsfactory to generate the [layout](https://gdsfactory.github.io/gdsfactory-photonics-training/notebooks/30_mzi.html#circuit-simulation)

it would also be able to programmatically create a symbol library fo the [gdsfactory PDKs
](https://gdsfactory.github.io/gdsfactory/components.html) and also allow to create complex schematics using python

StefanSchippers, 2024-02-19:
I have set up a test circuit:

[image]

and using the following tcl procedure:

```
proc sax_netlist {} {
  puts "\"instances\": \{"
  foreach {i s t} [xschem instance_list] {
    if { $t == {primitive} } {
      puts "  \"$i\": [file rootname $s],"
    }
  }
  puts "\},\n\"ports\": \{"
  set ports 1
  foreach i [xschem list_nets] {
    set net [lindex $i 0]
    set type [lindex $i 1]
    if {$type == {net} && $ports == 1} {
       set ports 0
       puts "\},\n\"connections\": \{"
    }
    if {$ports } {puts -nonewline "\"$net\":"}

    set first 1
    foreach c [xschem instances_to_net $net] {
      set inst [lindex $c 0]
      set pin [lindex $c 1]
      if {!$first} { puts -nonewline : }
      puts -nonewline " \"$inst,$pin\""
      set first 0
    }
    puts ","
  }
  puts "\},"
}
```
yields the following output:

```
"instances": {
  "lft": coupler,
  "top": waveguide,
  "rgt": coupler,
},
"ports": {
"out1": "rgt,out1",
"out0": "rgt,out0",
"in1": "lft,in0",
"in0": "lft,in1",
},
"connections": {
 "lft,out0": "rgt,in0",
 "lft,out1": "top,in0",
 "top,out0": "rgt,in1",
},
```
I think creating a YAML netlist is not diffcult.

joamatab, 2024-02-19:
amazing!

would it be possible to use python instead of tcl?

I've used python for the last 14 years but never used tcl before

StefanSchippers, 2024-02-19:
xschem is written in C and uses tcl-tk mainly for the GUI and scripting interface. (Python did not exist when i started xschem).

Anyway the idea is to have the small exporter procedure in TCL that xschem can load at startup, then you can call xschem from python to get the netlist.

something like: 
xschem test.sch --command 'source test.tcl; sax_netlist'
or just:
xschem test.sch --command 'sax_netlist' if the procedure is sourced on startup.

StefanSchippers, 2024-02-19:
You can also call xschem from python as a sub-process, send plain text commands to xschem's stdin and get responses from xschem'sstdout.

The above script can be entirely written in python, sending these 3 commands to xschem:

- `xschem instance_list`
- `xschem list_nets`
- `xschem instance_to_net <net>`

All commands are documented [here](https://xschem.sourceforge.io/stefan/xschem_man/developer_info.html#cmdref).
Example of xschem's output based on the sample circuit above.

`xschem instance_list` returns a list of `{instname} {symbol_name} {symbol_type} ...`
```
xschem [~/.xschem/xschem_library/sax_rf_examples] xschem instance_list
{lft} {coupler.sym} {primitive}
{top} {waveguide.sym} {primitive}
{rgt} {coupler.sym} {primitive}
{p4} {opin.sym} {opin}
{p5} {opin.sym} {opin}
{p2} {ipin.sym} {ipin}
{p6} {ipin.sym} {ipin}
```

`xschem list_nets` returns a list of {net_name type}, where type is either `ipin`, `iopin`, `opin` or `net`
depending if the net is attached to a circuit port or not.
```
xschem [~/.xschem/xschem_library/sax_rf_examples] xschem list_nets
{out1 opin}
{out0 opin}
{in1 ipin}
{in0 ipin}
{#net1 net}
{#net2 net}
{#net3 net}
```

`xschem instances_to_net <net>` returns as a list all instance pins attached to given net:
`{ {instance name} {pin_name} {X-coord} {Y-coord} } ... {..}` 
```
xschem [~/.xschem/xschem_library/sax_rf_examples] xschem instances_to_net {#net2}
{ {lft} {out1} {320} {-250} } { {top} {in0} {320} {-250} } 
```

StefanSchippers, 2024-09-05:
idle... closing for now...

closed 2024-09-05 by StefanSchippers