StefanSchippers/xschem #172: enable intuitive mode

joamatab, 2024-02-17:
i find xschem looks great, but is hard to get started if you have never used Cadence. For example i end up using Qucs-s because the UI is more intuitive

How could we enable an intutive mode? 

some examples on the intuitive behavior:

- you can move elements by dragging them with the mouse
- you can click on the pins and draw connections 

@proppy

StefanSchippers, 2024-02-17:
I think these 'intuitive' modes can be added. Will have a look on this.
For complex circuit design one ot the rules is that a circuit can not be changed by messing with the mouse, so randomly clicking and dragging in the circuit drawing area should not change anything. In many environments this is a de-facto rule. However nothing prevents us from using a couple of configure options to enable / disable these editing modes.

StefanSchippers, 2024-03-02:
I have added a so called "intuitive_interface" option. It can be set in the Options menu or permanently in xschemrc by setting `set intuitive_interface 1`

https://github.com/StefanSchippers/xschem/assets/69359491/a8fb5fa9-a1fe-4ace-9a82-75a05d9de9f1

joamatab, 2024-03-02:
Looks amazing! it's nice to be able to move the elements without breaking the connectivity. As a new user into schematic editors I was surprised this was not the default behavior

What are the default and new shortcuts?

A few more suggestions:

- [ ] add a help menu button that takes you to the [docs](http://repo.hu/projects/xschem/xschem_man/xschem_man.html)
- [ ] add in the help a cheat-sheet with all the default shortcuts, and the shortcuts for intuitive mode

StefanSchippers, 2024-03-03:
docs can be reached by going to Help-> About
[image]
The key shorcuts are in `Help->Keys`. I wil add some notes for Intuitive mode, however it's just point & click.
[image]

closed 2024-03-23 by StefanSchippers