StefanSchippers/xschem #173: Unable to launch Xschem

Aayushmi26muktui, 2024-02-18:
Hi,

I am in the last step but when I try to launch xschem,  it does not launch the xschem window and shows this.

[image]

Can you please help me ? What do you think I should do next ?

StefanSchippers, 2024-02-18:
Try to run any other linux graphical application, like xterm, xclock, xedit, 
If xschem does not open a window this means no Xserver is running. 
Try to see if DISPLAY variable is defined:
`echo $DISPLAY`
If nothing is returned try to set it up:
export DISPLAY=:0
but likely no Xserver is running.

Aayushmi26muktui, 2024-02-18:
Yess you are right no Xserver is running. How do it get Xserver running ?

I tried to install Xservers by :

sudo apt-get install xorg
sudo apt-get install openbox

But that does not fix my problem.

[image]

StefanSchippers, 2024-02-18:
If you are running a WSL linux under windows do some research. I think lot of people are using this setup, but I don't have a windows machine. 

Try this web page.
https://learn.microsoft.com/en-us/windows/wsl/tutorials/gui-apps

closed 2024-04-14 by StefanSchippers