StefanSchippers/xschem #175: [RFC] keep focus in search box in Choose Symbol window

qpwo, 2024-02-20:
Would be nice if the "File/Search" field was always focused (unless you are typing new dir I suppose)

[image]

(This would mirror behavior in some file explorers.)

Bonus request 1: If you're inserting symbol (rather than opening a file), then filtering to .sym should be the default.

Bonus request 2: The first result in the top-right could always be highlighted, so if you press enter then it's inserted (or the directory is opened)

StefanSchippers, 2024-02-20:
I have cleaned up the operations in the file chooser dialog box to allow keyboard only operation. See short clip.
https://github.com/StefanSchippers/xschem/assets/69359491/b9db65ea-a36b-412a-8415-3e0c77649949

qpwo, 2024-02-21:
Beautiful!

qpwo, 2024-02-21:
I swear it was working for me like your video a second ago, but now the file box is separate from the search box?? File box is focused -- maybe search box is supposed to be? I must have messed something up... 

[image]

StefanSchippers, 2024-02-21:
I separated the search pattern box from the file name box. 
So you can keep a search pattern and click files. Having these two things in one box was confusing.
The search box does not get focus with the TAB key, but you can click in it and set a pattern to restrict displayed files.

qpwo, 2024-02-22:
Personally I use the search box way more often. But I'm sure you have your reasons!

qpwo, 2024-02-22:
(Maybe you could drop the radio in the bottom left btw?)

closed 2024-02-22 by qpwo

StefanSchippers, 2024-02-22:
I can add the focus to the search when iterating with the TAb key. I thought it was disturbing. I am quite neutral about that :-)

StefanSchippers, 2024-02-22:
@qpwo I have done the change:
When opening the file load/insert dialog box the search entry is focused and text is selected.
Will also take focus when iterating with the TAB key.
Thank you.