StefanSchippers/xschem #177: Extended draw shapes capabilities

georgtree, 2024-02-25:
Hello!
It would be great to see the ability to draw more complex shapes - ellipse, Bézier curve, etc.
These shapes will greatly simplify component symbols creation, currently I can only approximate it with segmented line.
Also would be great to add scaling ability for already placed shapes.

Thank you in advance, George.

StefanSchippers, 2024-02-29:
I have added a first implementation of bezier curves. 
They are essentially a variation built on top of the polygon shape.
Draw a multipoint polygon (key=`p`) (npoints >= 3) , set attribute **bezier=1** on this shape, it will transform into a 
composite quadratic bezier spline. I have done tests with quadratic and cubic bezier splines. 
Quadratic beziers seem more useful, since they can be bent more easily than higher order polynomial curves.
I have to improve the UI for selecting and more easily moving the control points, but it is already useable. 
You can also transform a closed polygon into a bezier and in this case it can be filled (by adding attribute **fill=1**).

Attached the example schematic. Please do a test.
[image]
[test_bezier.zip](https://github.com/StefanSchippers/xschem/files/14442224/test_bezier.zip)

georgtree, 2024-02-29:
Wow, that was really fast, I tested it and all works great, thank you! 
I will play with it a little bit more!

StefanSchippers, 2024-02-29:
Thank you for testing.
I was a bit scared by the math, but turned out to be simpler than I thought.

For the ellips I will look at it. One important question is if you need X/Y oriented ellipses or any angle.

In the mean time I se beziers approximate ellipses quite well. Just start at the correct point as shown.
[image]

georgtree, 2024-02-29:
Hello, I need XY only for drawing pretty-looking symbols

StefanSchippers, 2024-02-29:
I have improved a bit the bezier editing. If you select one you can click and drag directly the control points. This applies also for normal polygons. Please test. Thanks!

Last thing I need to improve a bit is selection. Now selecting a bezier works well if you click one of the endpoints or close to a control point. I need to let user select the curve also by clicking any location close to the bezier line. 

[image]

georgtree, 2024-03-05:
Hello, thank you, it is very convenient and works as expected!

georgtree, 2024-03-30:
Hello!
Sorry for asking, but do you have any thoughts about the time the ellipsis will be available to draw? I am not pushing you just ask because I have a lot of symbols to modify and want to understand if it is worth to wait for proper ellipsis implementation instead of bezier approximation...
Thank you in advance!

georgtree, 2024-04-05:
Hello, all works great, thank you!
[image]

georgtree, 2024-04-05:
Hello, find the issue - when component is flipped or rotated, I get this behaviour:
[image]
[image]
Looks like the issue is in absolute definition of angles...

StefanSchippers, 2024-04-06:
Yes this will be fixed soon. Thanks

georgtree, 2024-04-06:
Hello, I tried with last commit, it works with ellipse that I place onto schematic, but if I generate symbol from tcl script, the problem is the same.

StefanSchippers, 2024-04-06:
Yes, bare ellipses are fixed now. Next commit will address symbols with
ellipses.

StefanSchippers, 2024-04-06:
Last commit hopefully does correct transformations for ellipses inside symbols. Please test.
Thank you for the feedback.

georgtree, 2024-04-06:
Hello!
Just tested, all works fine, thank you :)

closed 2024-04-06 by georgtree