StefanSchippers/xschem #178: improve color scheme

joamatab, 2024-03-02:
When using arch linux the edit simulators are barely readable
[image]

On debian i did not have that issue

StefanSchippers, 2024-03-02:
I don't know why that system decides to use white color for text. In 99% of the systems text is black unless changed by the user.
seems a dark theme colorscheme is used.
This is usually screwed up by some Desktop Environment.

StefanSchippers, 2024-03-02:
@joamatab I have added explicit setting of the following colors:
  option add *foreground black startupFile
  option add *activeforeground black startupFile
  option add *background {#d9d9d9} startupFile
  option add *activebackground {#ececec} startupFile
  option add *disabledforeground {#a3a3a3} startupFile

I will manage to make these options user selectable in xschemrc file.
For now it should avoid white text on light backgound, I hope.

StefanSchippers, 2024-03-06:
I have added  an option that can be set in xschemrc file:
`set dark_gui_colorscheme 1`
will start xschem with a dark colorscheme, where 
`set dark_gui_colorscheme 0 `
(the default) will use a light colorscheme.

this option affects the widgets, not the drawing area. A separate `dark_colorscheme` i used for the drawing canvas (default: set to 1), so 4 combinations are possible.

It is possible to customize all colors, but the risk of messing up things and having odd combinations is considerable.
[image]

joamatab, 2024-03-06:
amazing! thank you Stefan!

closed 2024-03-06 by joamatab