StefanSchippers/xschem #180: Issue with graph window

georgtree, 2024-03-05:
Hello!
After last update when I try to open graph window, it opens like this:
[image]
I use KDE Plasma with X11 windows manager.

StefanSchippers, 2024-03-06:
Can you resize the window and make it bigger?

StefanSchippers, 2024-03-06:
I have build xschem with tcl-tk 8.6 (the current version on my system) and with tcl-tk 8.4 (released 20 years ago) and it works.
[image]

georgtree, 2024-03-06:
Hello, yes it could be resized to work as usual manually.

georgtree, 2024-03-06:
Also after update I have display issues with Simulation Configuration window
[image]

StefanSchippers, 2024-03-06:
This is fixed in atest. I saw it yesterday.

StefanSchippers, 2024-03-06:
If the problem with graph dialog box showing up too small persists  I will force an initial size for it. Is that problem repeatable?
Thank you.

georgtree, 2024-03-06:
On my work machine graph window is ok, will test it again on home machine

StefanSchippers, 2024-03-06:
Thank you. In the mean time I wanted this dialog box to "remember its size" so if user resizes it the next time it is opened it will use the last size. (in the running session). so I have to add this function anyway.

closed 2024-03-23 by StefanSchippers