StefanSchippers/xschem #183: How to change default web browser

mzannoni, 2024-03-15:
I'm on Manjaro linux and I've seen that links in scheamtics can't be opened, because the command `x-www-browser` can't be executed.

I launch the example schematic and then try and `Ctrl+<left-click> on a link and:  
```
❯ xschem .local/share/doc/xschem/examples/0_examples_top.sch
% tclvareval(): error executing launcher {http://repo.hu/projects/xschem} {x-www-browser}: couldn't execute "x-www-browser": no such file or directory
```

Is there a way, e.g. in `xschemrc` to specify the browser? (could't find it)

Or more in general, any help in making this work would be welcome :wink:

StefanSchippers, 2024-03-15:
Try to remove the `program=x-www-browser` by clicking on the symbol, pressing 'q' and deleting the `program=...` line.
If it still does not work add `program=firefox` or any other browser installed on the system.

Linux desktops usually set `x-www-browser` as a link to the default installed browser, so whether it is chrome, firefox or anything you can call it in a portable way. On some systems this method is broken.

StefanSchippers, 2024-03-15:
I have removed the `program=...` lines in the launchers in the `0_examples_top.sch`, since if no `program=...` is specified xschem uses `xdg-open`, another standard desktop command to open a file or URL with the right application:

xdg-open image.jpg --> uses the default image viewer
xdg-open file.pdf --> uses the default pdf viewer
xdg-open dile.txt --> uses the default editor
xdg-open https://www.google.com --> will use the default browser

If no `program=...` is specified in the launcher xschem uses the value specified in `launcher_default_program` tcl variable, which is by default `xdg-open`. This default file open application can be set and changed in xschemrc:
`set launcher_default_program ....` 
however it must be something that recognizes different file formats, like xdg-open, so unless you know what you are doing don't change it.

closed 2024-03-23 by StefanSchippers