StefanSchippers/xschem #184: Running into library dependency issue even though it is installed.

ddosluuw, 2024-03-23:
Hello, when I run configure I receive the following message: 

_Checking for tk... 8.6... (no tcl) 8.5... (no tcl) 8.4... (no tcl) plain... (no tcl) not found
Node libs/script/tk/* is required but provided detection callback fails to find that feature on that system._

However, tcl is already installed:

_sudo yum list tcl
Loaded plugins: fastestmirror, langpacks
Loading mirror speeds from cached hostfile
 * base: or-mirror.iwebfusion.net
 * centos-sclo-rh: ix-denver.mm.fcix.net
 * centos-sclo-sclo: mirrors.vcea.wsu.edu
 * epel: ziply.mm.fcix.net
 * extras: ziply.mm.fcix.net
 * updates: mirrors.rcs.alaska.edu
Installed Packages
tcl.x86_64                                                     1:8.5.13-8.el7                                                     @base
Available Packages
tcl.i686                                                       1:8.5.13-8.el7_ 

Has anyone run into this issue with the configure script?

ddosluuw, 2024-03-23:
I am installing this on Centos 7.

StefanSchippers, 2024-03-23:
you need the development packages, these contain the files needed to build programs that use tcl and tk
usually these packages are called tcl8.6-dev, tk8.6-dev or similar, depending on the distribution.
for centOS probably tcl-devel-8.5.13-8.el7.x86_64.rpm and tk-devel-8.5.13-6.el7.x86_64.rpm

closed 2024-03-24 by ddosluuw