StefanSchippers/xschem #186: Meaning of "+" and "|" in .sch files

urish, 2024-03-25:
Sometimes, there are + or | symbols in the middle of the attr list, e.g.:

https://github.com/StefanSchippers/xschem/blob/aa6182c65f1b0379e13f7180f67e9f7e2dceb9f9/xschem_library/rom8k/rom2_predec3.sch#L128-L130

or

https://github.com/vincentfusco/tt06_555/blob/d58ca5dc6e58d8d6e327737296cf200e120c4b07/xsch/ip/timers/timer_core.sch#L188-L189

What is the meaning of those symbols?

StefanSchippers, 2024-03-25:
Hi, I think the `|` in the second case is a typo. 
The `+` in the lvnot.sym is a "spice netlist cognitive bias"
+ is the character to use when you break a long line in a spice netlist.
In this context it is just useless, I think I will remove it in the example circuits.
An attribute list is just a string with key=value assignments. Anything else is allowed, if it has not the form key=value it is ignored.

urish, 2024-03-25:
> An attribute list is just a string with key=value assignments. Anything else is allowed, if it has not the form key=value it is ignored

Thanks! And can the keys have any characters other than } and whitespace? or is there a limited set of characters valid for a key?

StefanSchippers, 2024-03-25:
keys can have any characters except newline, backslash, double quote (") , space, tab and ;
the { character is allowed (hopefully no one will use it) xschem will escape it when puttintg the attribute list inside {...}

urish, 2024-03-25:
Thanks! The xschem-viewer will now happily accept those files :)

closed 2024-03-25 by urish

StefanSchippers, 2024-03-25:
You are welcome. If you find other unclear things just ask!