StefanSchippers/xschem #188: i can not see the symbols when opening an example

princeTem, 2024-03-30:
Dear Stefan
i use a standalone ubuntu system , i followed the steps as your video of installation , but when i open the example , i found the following ,, i tried every possible way to find a solution 
can you help me with that 
[image]

StefanSchippers, 2024-03-30:
what commands did you run  exactly?

go into the source xschem directory and do again these commands:

./configure 
make
sudo make install

princeTem, 2024-03-30:
dear Stefan 
thank you , it worked after i removed the previous installation ,,, and i installed it from beginning , then i used the entry ./configrue  without the " prefix" 
it seems that at the first installation , i used the prefix with a directory which caused me trouble as i am new on ubuntu 
thanks alot again 
regards

closed 2024-04-14 by StefanSchippers