StefanSchippers/xschem #191: Issue with external data .csv plotting

georgtree, 2024-04-07:
Hello!
I found a possible bug in loading data from external .csv file: if in the name of column the upper case letter exists,
the graph corresponding to that column isn't shown in graph window.
[ElasticBearing.csv](https://github.com/StefanSchippers/xschem/files/14897849/ElasticBearing.csv)

In attachment I have issues with idealGear.flange_a.tau column, after changing to idealgear.flange_a.tau all works well.

StefanSchippers, 2024-04-07:
George, thank you for reporting this. 
I think this is now fixed. xschem did the node name look up either all lower case or all uppercase, lookup is now done on the node name as is before trying lowercase / uppercase.

These upper/lower conversions are done because ngspice implicitly converts all circuit node names to lowercase, while Xyce does convert everything to uppercase.
so xschemnode `Load` becomes `load` in ngspice and `LOAD` in Xyce. to avoid users getting mad trying to give the accepted name when plotting data xschem tries all these combinations.

georgtree, 2024-04-08:
Hello, just tested, all works well, thank you!