StefanSchippers/xschem #192: RFC: QR codes??

qpwo, 2024-04-09:
This is probably a bad idea but curious for your thoughts. Feel free to close without comment as always.

Problem: Screenshots in bug reports and tutorials can be hard to reproduce because linking/including the `.sch` file is a bit of a pain. Then when you update your bug report or tutorial you have to make sure the implied `sch<->png` correspondence is still correct. Basically, keeping two files synchronized is difficult.

Solution:

* Add qr_code.sym. It always encodes the current state of the `.sch` file. So you can include it in your screenshots for your tutorials and so on.
  * (Maybe the qr code gets a red outline if there's unsaved changes?)
* The "open file" dialogue can show pngs in addition to `.sch` and `.sym`. Or maybe you drag and drop an image into the xschem window? If there's no QR code then you get an error. Otherwise, the QR code is parsed and the circuit is opened in a new untitled tab.

StefanSchippers, 2024-05-21:
if a schematic is in a public repository like github instead of sending images you can open directly with xschem locally:
`xschem  https://raw.githubusercontent.com/w32agobot/SKY130_SAR-ADC/main/xschem/adc_top_tb.sch`

So just provide the URL of the schematic, xschem can view it. If you associate .sch files with xschem in your desktop may be you can just click the URL and get it opened by xschem (however never tried, as I don't even use a desktop environment).

There is also a good xschem online viewer now https://xschem-viewer.com/

qpwo, 2024-05-22:
Cool yeah QR code thing probably silly. Thanks for tip.

closed 2024-05-22 by qpwo