StefanSchippers/xschem #193: Ability to rotate/flip during "xschem place_symbol" command

georgtree, 2024-04-14:
Hello!
It would be a nice feature to have an ability to rotate/flip symbols during `xschem place_symbol` command, now it is necessary place symbol first and only then we can rotate it.
Thank you in advance.

StefanSchippers, 2024-04-14:
While placing the symbol you can press Shift-R to rotate of Shift-F / Shift-V to flip the component before clicking the mouse to complete the placement.

If you want a complete programmatic way to place a symbol (no user interactive . gui action) see the `xschem instance` command.

georgtree, 2024-04-14:
It is strange, but I can't do it not from component browser nor from my own palette with this command...is there any switches that prevent this?

StefanSchippers, 2024-04-14:
Press Insert to bring up the component browser, click one component, in the drawing window press Shift-F or Shift-R to flip / rotate.

If that does not work you are probably unable to flip / rotate any component in a schematic and this is strange.

https://github.com/StefanSchippers/xschem/assets/69359491/6b8d4f12-f8c3-47b4-9bdd-b79422191ae5

georgtree, 2024-04-14:
It works from insert, but not from the component browser window, or when I call procedure from my own gui

StefanSchippers, 2024-04-14:
What are *exactly* the operations you do to bring up the component browser and the following actions?

georgtree, 2024-04-14:
Open component browser -> select component there -> bring component to the xschem window -> press shift+R

georgtree, 2024-04-14:
Ah, I found an issue - there was no focus at xschem window, I have to bring focus back by pressing to main window frame...

closed 2024-04-14 by georgtree

StefanSchippers, 2024-04-14:
Ah ok, There is a setting in xschemrc:
 `set autofocus_mainwindow 1`
(if you don't have this setting it is already enabled)
that is supposed to focus the main window when entering it. However this does not work universally as the window manager is the ultimate culprit when there are focus issues.