StefanSchippers/xschem #194: Preview symbol command issue

georgtree, 2024-04-14:
Hello!
I tried to generate preview of the symbol with preview_window command:
`toplevel .symwin -width 200 -height 200
xschem preview_window create .symwin
xschem preview_window draw .symwin /usr/local/share/xschem/xschem_library/devices/cccs.sym`
But it doesn't show anything:
[image]
No error in terminal.
Thank you in advance!

StefanSchippers, 2024-04-14:
try this:
```
toplevel .symwin -width 200 -height 200 -bg {}
xschem preview_window create .symwin
xschem preview_window draw .symwin /usr/local/share/xschem/xschem_library/devices/cccs.sym
bind .symwin <Expose> {xschem preview_window draw .symwin /usr/local/share/xschem/xschem_library/devices/cccs.sym}
```
the toplevel is created without a background, otherwise there is a race condition between xschem drawing the circuit in it and tk painting the background.

The bind command ensures the window will be refreshed if it is obscured / iconified

StefanSchippers, 2024-04-14:
I have made some updates so that previous commands can be simplied as:
```
toplevel .symwin -width 200 -height 200 -bg {}
xschem preview_window create .symwin
xschem preview_window draw .symwin /usr/local/share/xschem/xschem_library/devices/cccs.sym
bind .symwin <Expose> {xschem preview_window draw .symwin}
```

georgtree, 2024-04-14:
All works great, thank you!

A side question - how to add to xschemrc file a `source` of script that contains `xschem` command?
It tells me that:
[image]
Looks like I have to load some xschem package before the loading of my script...

StefanSchippers, 2024-04-14:
you need to use this:
` set postinit_commands {source /path/to/file}`

georgtree, 2024-04-15:
Hello!
I tried add this command to run after init, but it doesn't work:
`set postinit_commands {source /home/george/MDPSBLib/gui_plugin/component_palette_grid.tcl}`
I added puts command there but it doesn't show at startap, and at the same time the default command 
`set postinit_commands {puts [xschem get version]}` works as it should.
Maybe I should add something to my script?

georgtree, 2024-04-15:
I add this:
`frame $paletteFrame.symbPreview -bg {}
bind $paletteFrame.symbPreview <Expose> {xschem preview_window draw $paletteFrame.symbPreview}
xschem preview_window create $paletteFrame.symbPreview`

And got segfault at script run, without this line:
`bind $paletteFrame.symbPreview <Expose> {xschem preview_window draw $paletteFrame.symbPreview}`
window works but I have distorted view when drag top window which contains preview window:
[image]

georgtree, 2024-04-15:
Now my GUI looks like this:
[image]
still have segfault when I try to bind...

georgtree, 2024-04-17:
I added 
`ttk::style configure TLabelframe -transparentcolor`
but it doesn't help :(

georgtree, 2024-04-17:
I found the issue, I am an idiot, sorry

closed 2024-04-17 by georgtree