StefanSchippers/xschem #195: Preview .tcl symbol with preview_window

georgtree, 2024-04-17:
Hello!
Question - is it possible to preview generator symbol with command `preview_window` with default input for generator?
It should be possible somehow, but I need your advice, thank you.

StefanSchippers, 2024-04-18:
it is possible, to render a generator you must call its name with parenthesis, perhaps with arguments. Since in a preview window you are not usually giving any parameters the generator script should be usually architected to accept no arguments and do a "default" action. 

To preview a generator tier.tcl:
xschem preview_window draw .p {xschem-repo/trunk/xschem_library/generators/tier.tcl()}

[image]

georgtree, 2024-04-18:
Great, thank you, I have many components that use scripts not for changing the symbol but it's properties in device_model attribute. I will try it today's evening.

georgtree, 2024-04-18:
Hello, don't understand why, but the else statement doesn't work, if tag is 'element', the preview window works as expected, if not, nothing happens, even `puts` doesn't output anything to console. Maybe you know why is it so? 
`    set focusTag [$pf.palettePanwin.comp.tree item [$pf.palettePanwin.comp.tree focus] -tag]
    if {$focusTag eq {element}} {
        puts 1
        $pf.palettePanwin.prev configure -background {}
        xschem preview_window draw $pf.palettePanwin.prev [$pf.palettePanwin.comp.tree focus]
        bind $pf.palettePanwin.prev <Expose> {xschem preview_window draw $pf.palettePanwin.prev }
        bind $pf.palettePanwin.prev <Configure> {xschem preview_window draw $pf.palettePanwin.prev }
    } else {
        puts 2
        $pf.palettePanwin.prev configure -background {}
        xschem preview_window draw $pf.palettePanwin.prev "[$pf.palettePanwin.comp.tree focus]()"
        bind $pf.palettePanwin.prev <Expose> {xschem preview_window draw $pf.palettePanwin.prev }
        bind $pf.palettePanwin.prev <Configure> {xschem preview_window draw $pf.palettePanwin.prev }
    }`

georgtree, 2024-04-18:
All works, thank you!
Additional question - is it possible to show hidden by default text in preview window?
Thank you in advance!

StefanSchippers, 2024-04-18:
I think you can temporarily enable the global variable **show_hidden_texts**:

```
set save $show_hidden_texts
set show_hidden_texts 1
xschem preview_window draw .path /path/to/symbol.sym
set show_hidden_texts $save
```

georgtree, 2024-04-18:
Ah, right, thank you!

closed 2024-04-18 by georgtree