StefanSchippers/xschem #196: Issue with previe_window

georgtree, 2024-04-19:
Hello!
Could you help please - I have preview_window in detached window, all works, but after I make **Save As** action in main window, preview_window command stops work until I kill current window and launch tcl script again. I draw command in event handler, so after window stops work, event handler still work. 
What could be an issue?
I add a video.
Thank you in advance.

https://github.com/StefanSchippers/xschem/assets/113975847/f93b25d3-1f96-4ab6-887f-fceb99d72be6

StefanSchippers, 2024-04-19:
Yes I need to address this. The reason is simple. The preview_window() function (in xinit.c) 
saves as static (inside the function) data the informations about the previewed schematic.
This means only one preview at a time may be active. Opening a second preview instance will overwrite the data of the previous preview and likely cause a lot of trouble.

georgtree, 2024-04-19:
Is it possible to return working state of preview window?

georgtree, 2024-04-19:
I solved this issue by recreate preview_window each time in event handler

StefanSchippers, 2024-04-19:
This is a possibility, but I am working on a preview functions that allows multiple previews at the same time...

StefanSchippers, 2024-04-19:
Ok, now preview_window should allow up to 10 open previews.
You can do something like:
```
toplevel .a -bg {}
toplevel .b -bg {}
xschem preview_window create .a 
xschem preview_window create .b
xschem preview_window draw .a xschem-repo/trunk/xschem_library/examples/poweramp.sch
xschem preview_window draw .b xschem-repo/trunk/xschem_library/examples/test_lm324.sch
```
You can repeat a `xschem preview_window draw .a another_schematic` to update .a with a different schematic.
or you can do a 
```
xschem preview_window destroy .a
...
...
toplevel .a -bg {}
xschem preview_window create .a
xschem preview_window draw .a yet_another_schematic
```

georgtree, 2024-04-19:
Yes, just tested, all work !

closed 2024-04-19 by georgtree