StefanSchippers/xschem #197: Highlight nets connected to particular custom type of pin

georgtree, 2024-04-22:
Hello!
I have a question - is it possible to define custom type of the pin, for example, electrical type, magnetic, thermal, signal, etc, and when wire is connected to this pin, it is highlighted with colour that correspond to that type of net? For example, we connect wire to pin with thermal type, and it turns red, and if we accidentally connect it to other type, it turns grey to show that type are mismatched.
By this I want to highlight connections inside particular physical domains, so we can visually diagnose if pins with different types are connected, and see which type of pins we connect with this particular wire.
Thank you in advance.

StefanSchippers, 2024-04-24:
Hi, George,

net / pin types are already used for VHDL. 
a sig_type attribute is attached to a net / pin to define the type of the signal, since VHDL allows any type (primitive or user defined), like bit, std_logic, real, integer and so on.
[image]

I think you can reuse this attribute. There is also a `class=...` attribute that defines the class of the signal, for VHDL this could be for example `signal`, `constant`. I think class can be used to define the class of the signal (electrical, thermal, mechanical,...) and the sig_type should define the representation (bit, real, integer)....

Then some scropt needs to be realized to do what you want. 
If you can send a simple working example (with the sub-schematics/sub symbols) I can do some experiments.

georgtree, 2024-05-19:
Hi, sorry for late reply - by working example, what do you mean? the example that can be run sucesfully in simulator, or just example of xschem .sch file? If the second, I can send it to you right now, the only thing that I don't know how to do is - save all elements locally with full hierarchy.

StefanSchippers, 2024-05-20:
You can send a cut-down example of your design containing two or more signal domains.
Create a new empty directory (example/), place the schematics and symbols used into this directory.
set XSCHEM_LIBRARY_PATH in your xschemrc to contain only xschem base devices and this directory:
`set XSCHEM_LIBRARY_PATH ${XSCHEM_SHAREDIR}/xschem_library/devices:/path/to/your/example`
Verify xschem can open the example and there are no missing symbols.
Then zip the directory so I can run the test.

georgtree, 2024-05-20:
Hello!
I did it, hopes all will be ok.
Attached file to this post,  I added three classes to pins - magnetic, mech_trans and electrical.
Also I thing that subcircuit pins should inherit the class from device at lower level.
Thank you in advance.
[example.zip](https://github.com/StefanSchippers/xschem/files/15380688/example.zip)

StefanSchippers, 2024-05-21:
I have created the following script: 
```
proc hilight {} {
  set color(magnetic)   0 ;# red
  set color(electrical) 1 ;# yellow
  set color(mech_trans) 2 ;# white
  
  foreach item [xschem list_nets] {
    lassign $item net type ;# type can be net or ipin or iopin or opin
    set class {} 
    foreach item [xschem instances_to_net $net] {
      lassign $item i p x y  ;# i=instname, p=pinname, x and y are unused coorditanes
      set cl [xschem getprop instance_pin $i $p class]
      if { $class ne {} && $cl ne $class } { ;# print error if a net connects to different class domains
        puts "error: $net connects to $cl and $class"
      }
      set class $cl
    }
    puts "$net --> $class"
    xschem set hilight_color $color($class)
    xschem hilight_netname $net
  }
} 

hilight
```
Applying the script to your schematic yields this:
[image]

where red is magnetic, yellow is electrical, white is mech_trans.
I had to change the "mechanical ground" label to GND instead of 0 otherwise the same label (0) connects to different class domains.

georgtree, 2024-05-21:
Wow, that is super fast and cool, I will try it out this evening!
One question - globally, all grounds should be connected to 0, but to make this highlighting tool work properly we need different labels.
How to set different labels to be connected to global ground during netlisting? 
The way I see is define .subckt for each of the ground symbol and connect them to zero explicitly inside subcircuit, but maybe there is more elegant way to do this, some alias for example?

StefanSchippers, 2024-05-21:
It is not clear to me how an electrical ground (like 0) relates to a "magnetic" ground. If they are different things they should have different labels. 

In your ground symbols there is the `global=true` attribute. In the spice netlist a `.global net` line is produced for each ground label (example: `.global GND`)

eEach global label has global scope. Anywhere in the hierarchy they refer to the same net.

georgtree, 2024-05-21:
For simulator they are identical, because ngspice doesn't know about any domain, it is an abstraction that help to work with different circuit variables like current-voltage, angular speed-torque, flux-mmf etc.
When circuit is netlisted to ngspice, all grounds must be united.

StefanSchippers, 2024-05-21:
Ok, understood. So you can modify the script to not consider the '0' node (it can be connected to any domain, so highlighting it does not make sense).

StefanSchippers, 2024-05-21:
For in, out, inout pins of a schematic, add the **class=...** attribute, as well as to the corresponding pins in the symbol.
I prefer the user takes the task to assign class attributes to pins, instead of propagating these automatically.
This way the user has more control, and the script can flag unintentional domain crossing of nets.
Once you highlight nets in a subcircuit if you go up in the hierarchy the highlihts are propagated to parent circuit.

georgtree, 2024-05-28:
Hello!
Is it possible to define aliases to 0 node, that will be replaced during netlist generation?
I tried to use shorts with zero-voltage sources, but it affects the convergence of circuits...

georgtree, 2024-05-28:
The idea is to have different common nets for each domain to highlight accordingly (elec_ref, mech_ref, etc), but in netlist these nets must be replaced by the 0 reference. 
So, in .sch file we see them as different nets, but make a substitution during netlisting.

georgtree, 2024-06-20:
> Hello! Is it possible to define aliases to 0 node, that will be replaced during netlist generation? I tried to use shorts with zero-voltage sources, but it affects the convergence of circuits...

Hello! Is it possible to do something like this? Make a substitution of particular net names during netlisting

StefanSchippers, 2024-06-23:
the 0-voltage sources (ammeter) are the best way to create net aliases. One caveat: the 0 node is by definition global, so it has the same meaning in every subcircuit without needing to propagate the 0 node with pins. Any other net name (excluding GND which is equivalent to 0 in ngspice) is not global. So if you create another ground alias with a 0V voltage source you must set the node to global. This can be done by using the `vdd.sym` or `gnd.sym` symbols instead of a plain net label, These symbols add the `.global` declaration to the generated spice netlist making the aliases global.
[image]

see the generated netlist of above example:
```
** sch_path: /home/schippes/untitled.sch
**.subckt untitled
Vmeas 0 GNDA 0
.save i(vmeas)
Vmeas1 0 GNDB 0
.save i(vmeas1)
**.ends
.GLOBAL GNDA
.GLOBAL GNDB
.end
```

georgtree, 2024-06-23:
Hello, yes I tried this, but it affects convergence of some circuits, and in random way, I can't find the pattern to predict that.

georgtree, 2024-06-23:
I have an idea - i can add my own script between netlist generated by scheme and ngspice, directly in command to simulator, what do you think?

StefanSchippers, 2024-06-23:
Of course you can. Still curious why a 0V vsource for a net aliasing makes simulation fail.

georgtree, 2024-06-23:
Me too, I suspect that it adds additional entry in mna matrix, and makes matrix too stiff because of zero series resistance, but I am note sure.

StefanSchippers, 2024-06-23:
Can you try to alias a gnd node by usong a 0.000001 Ohm resistor?

StefanSchippers, 2024-06-23:
you can posprocess the netlist, transforming all:
Vxxxx node1 node2 0
to
RVxxxx node1 node2 0.000001
and see if simulation works.
No need to tamper the schematic of course.

georgtree, 2024-06-23:
Ah, you are right, I can postprocess netlists, thank you, I will check that

closed 2024-08-06 by StefanSchippers