StefanSchippers/xschem #198: tcleval(): evaluation of script: fet_drc failed

sfs325, 2024-04-24:
Dear all, I recently installed Xschem version 3.4.5. I have started with an example with a simple mosfet and two voltage supplies to check the operation point. As soon as I try to change any parameter of the mosfet I get the following error:

tcleval(): evaluation of script: fet_drc {M1} {/usr/local/share/pdk/sky130A/libs.tech/xschem/sky130_fd_pr/nfet3_01v8.sym} {nfet_01v8} {20} {0.15} {1} failed
         : invalid command name "fet_drc"

StefanSchippers, 2024-04-24:
This error does ot prevent correct netlisting. fet_drc is a drc checker for transistor dimensions. The drc checker (fet_drc_ is included in recent [xschemrc](https://github.com/StefanSchippers/xschem_sky130/blob/main/xschemrc) in the [xschem_sky130](https://github.com/StefanSchippers/xschem_sky130/tree/main) repository.

To fix add this procedure at the bottom of your xschemrc file:
```
# sky130 mosfets dimension checks
proc xfet_drc {instance symbol model w l nf} {
  set res {}
  # puts "$instance $model $symbol w=$w l=$l nf=$nf"
  if { [string is double $w] && [string is double $l] && [string is integer $nf]} {
    # not *_nf.sym devices: get single finger width
    if {![regexp {fet.*_nf\.sym$} $symbol]} {
      set w [expr { double($w) / double($nf)}]
    }
    switch -regexp $model {
      {[np]fet_01v8$} {
        if { $w < 0.42 } {
          append res "${instance} ($model): finger width is too small, w / nf = $w" \n
        }
        if { $l < 0.15 } {
          append res "${instance} ($model): length is too small, l = $l" \n
        }
      }
      {pfet_01v8_lvt$} {
        if { $w < 0.42 } {
          append res "${instance} ($model): finger width is too small, w / nf = $w" \n
        }
        if { $l < 0.35 } {
          append res "${instance} ($model): length is too small, l = $l" \n
        }
      }
      {nfet_01v8_lvt$} {
        if { $w < 0.42 } {
          append res "${instance} ($model): finger width is too small, w / nf = $w" \n
        }
        if { $l < 0.15 } {
          append res "${instance} ($model): length is too small, l = $l" \n
        }
      }
      {[np]fet_g5v0d10v5$} {
        if { $w < 0.42 } {
          append res "${instance} ($model): finger width is too small, w / nf = $w" \n
        }
        if { $l < 0.5 } {
          append res "${instance} ($model): length is too small, l = $l" \n
        }
      }
      {pfet_g5v0d16v0$} {
        if { $w < 5 } {
          append res "${instance} ($model): finger width is too small, w / nf = $w" \n
        }
        if { $l < 0.66 } {
          append res "${instance} ($model): length is too small, l = $l" \n
        }
      }
      {nfet_g5v0d16v0$} {
        if { $w < 5 } {
          append res "${instance} ($model): finger width is too small, w / nf = $w" \n
        }
        if { $l < 0.7 } {
          append res "${instance} ($model): length is too small, l = $l" \n
        }
      }
    } ;# switch
  }
  return $res
}

```

sfs325, 2024-04-24:
Thank you for your reply. I see that it does not prevent from netlisting. I see the procedure in fet_drc xschemrc. Unfortunetly your fix does not work. At least I have reported it.

StefanSchippers, 2024-04-25:
Ensure you have added the **fet_drc** procedure in the **xschemrc** you are using. Xschem loads first the **xschemrc** in the directory where it is started, then if no xschemrc is found it loads the one in your home directory: **~/.xschem/xschemrc**.

If this still fails ensure you have updated version of the xschem symbols. Unfortunately this drc check was asked a couple weeks ago, and to make it work with no problems all the tools must be updated. You can copy all the sky130 symbols from [here](https://github.com/StefanSchippers/xschem_sky130/tree/main/sky130_fd_pr) in your pdk installation  (usually `../share/pdk/sky130A/libs.tech/xschem/sky130_fd_pr`)

closed 2024-05-21 by StefanSchippers