StefanSchippers/xschem #204: Feature - add ability to set custom bitmap inplace

georgtree, 2024-05-18:
Hello!
It would be great to have feature like this - in property of particular polygon/rectangle we would able to set the custom bitmap of filling pattern. I use different colors for symbols to differentiate physical domains (mechanical, magnetic, etc), and it would be great to have an ability to visually differentiate polygons by different fill pattern with the same layer used.
Thank you in advance!

StefanSchippers, 2024-05-18:
When xschem was developed 20+ years ago, Xservers were slow if many shapes were drawn each with different fill patterns or other drawing attributes. For this reason a set of layers was created each with different color  / fill pattern, so if 200 shapes on layer 3 are to be drawn there is no Xserver call (XSetFillRule, XSetFillStyle, XSetStipple, XSetTile and so on) for each shape. All 200 shapes are drawn with the same graphic context (GC) before changing and moving to the next layer.

Nowadays this is probably not more a big issue, so may be some attributes my be defined fo specify a custom fill. In the future when graphic calls will be replaced by non X11 calls also an alpha value may be defined for transparency. This will require some work however.

StefanSchippers, 2024-05-20:
One more question. Currently fill pattern in xschem are tiles of 16x16 pixel squares.
the content of the 16x16 pixel can be defined by user, but how would you specify a fill pattern? by carefully providing these 16x16 bitmaps you can generate various patterns, for example:
[image]
but you must design the bitmap carefully so tiled patterns are uniform.

georgtree, 2024-05-20:
Hello, yes, I see, thank you for this notice

StefanSchippers, 2024-05-21:
As of now all xschem (closed) shapes support the fill attribute:
fill=0 : no fill
fill=1 : use the default pattern defined for that layer
fill=full: use solid fill

I will eventually add some other option to the fill attribute to set a custom pattern. I need to figure out a user friendly way to do it.

georgtree, 2024-05-22:
Thank you, for my purpose it is more than enough !

closed 2024-05-22 by georgtree