StefanSchippers/xschem #205: TAB symbol displaying issue

georgtree, 2024-05-21:
Hello!
I found the issue with TAB symbol in text block:
[image]
It renders it like a square, for now I use four spaces for indentation, but maybe there is a quick fix for it?
Thank you in advance!

StefanSchippers, 2024-05-21:
Yes, TAB is not handled (yet) so you get a weird 'unprintable char' placeholder.
TAB is not a character to display, but an information to use to set the position of the next character.
I will eventually handle that.

StefanSchippers, 2024-05-22:
George, 
can you please update xschem and test TABs in your examples? 
It should work now.
tcl variable `tabstop` sets the tab position (default 8) 
you can set this in xschemrc:
`set tabstop 4`
to change the default tab stops.

georgtree, 2024-05-24:
Hello, all works, thank you!

closed 2024-05-24 by georgtree