StefanSchippers/xschem #206: How to get value of cursor A

georgtree, 2024-05-21:
Hello!
I was able to print cursor B value by [xschem raw value rdson_vec {} -1] in tcleval block, but how to get cursor's A value?
Or distance between cursors (I suspect we can do by reading global variables cursor1_x and cursor2_x)? it would be great to have an ability to do additional calculation like frequency calculation, or absolute delta of y.
Thank you in advance.

StefanSchippers, 2024-05-21:
you can get cursor positions with these commands:
```
xschem [~] xschem get cursor1_x
0.0232642
xschem [~] xschem get cursor2_x
0.0248698
xschem [~] 
```

You can get vectors with a loop like this (**outm** variable vs **time**):
```
foreach i [xschem raw values time] j [xschem raw values outm] {
  puts "$i --> $j"
}
```
You can then break the loop if the sweep variable (time in the example) crosses the cursor position and print the variable value.
You can not immediately get the variable value given the cursor, since in general (for transient simulations) the sweep axis (time) has non uniform increments. You must inspect the vector elements.

StefanSchippers, 2024-05-21:
Another possibility is to set the cursor b value to a given x-axis point:
**xschem set cursor2_x 0.025**
and then get all variable values at the cursor B position.:
**xschem raw value rdson_vec {} -1**

StefanSchippers, 2024-05-21:
Following commands are added to xschem (update your xschem installation)
`xschem swap_cursors`: swap A and B cursor positions.
`xschem cursor n e` enable / disable cursors: example:
  `xschem cursor 1 1` (enable cursor 1)
  `xschem cursor 2 0` (disable cursor 2)

If a raw file is loaded, you can do:
```
xschem [~] xschem cursor 2 1
xschem [~] xschem set cursor2_x 0.024
xschem [~] xschem raw value outm {} -1
-3.8386496
```
to enable the b cursor, set it to some X value and then get variable values at cursor position.

georgtree, 2024-05-22:
Hello, thank you for suggestion and new commands!

closed 2024-05-22 by georgtree