StefanSchippers/xschem #207: Different color for currents and voltages in back-annotation

qpwo, 2024-05-25:
I think the annotations would be easier to read if the currents were in a different color. It can get crowded.

[image]

StefanSchippers, 2024-06-21:
This can now be done in two ways:
1) change the symbol and move the current / voltage text on a different layer 
2) on a specific symbol instance add attribute text_layer_n = lay, where n is the text number in the symbol (first=0, second =1, and so on, you can see the text position in the symbol by clicking on it and pressing Shift-S)
example: text_size_4=9  --> move the symbol fourth text item to layer 9

The second solution does not require to change the symbols library, but needs the above attribute on all instances where text must be rendered on a different layer.

StefanSchippers, 2024-06-21:
I have modified the devices/ xschem base symbols to display current annotations (@spice_get_current) on layer 12 (blue)
[image]

qpwo, 2024-06-21:
Awesome! Would've taken synopsys a decade 

[image]

closed 2024-06-21 by qpwo