StefanSchippers/xschem #208: Improve annotation magnitude clarity

qpwo, 2024-05-25:
When looking at currents in an annotated circuit, `843.76f` and `843.76n` look rather similar. So do `843.76f` and `8.4376f`. Sometimes I make mistakes because of this. I wonder if there's some easy improvement to make? A few ideas, best ideas first:

1. make bigger numbers a bigger font
2. add option to use fixed precision. If you set it to `e-11` then the number would always render like `XXX.XXn`. I guess it would be separate options for currents and voltages. Or maybe only an option for currents.
3. make the f,n,u, etc different colors
4. add an option for minimum current to show in annotation

I think any one of these would be sufficient and a great improvement

StefanSchippers, 2024-06-11:
You can make texts bigger by adding attribute `text_size_n=s`, where `n` is an integer representing the text item in the symbol: 0= first text item, 1= second text item and so on, s is a double representing the desired text size.
In a `lab_pin.sym` label adding text_size_1=0.5 will make the annotated voltage value bigger than default (0.2)

StefanSchippers, 2024-06-23:
This is an example where both label and annotated voltage are set to bigger size. See the `text_size_* = 0.7` attributes:
[image]

closed 2024-06-23 by StefanSchippers