StefanSchippers/xschem #212: ./configure: not found

Camisad0, 2024-05-29:
Hi,

I'm encountering error during the ./configure part when installing xschem.

Kindly refer to the attached image.
[image]

Looking forward to your inputs.

Thanks!

Camisad0, 2024-05-29:
Libraries checked 
[image]

StefanSchippers, 2024-05-29:
install all the packages listed [here](https://xschem.sourceforge.io/stefan/xschem_man/install_xschem.html) and run ./configure again.

StefanSchippers, 2024-05-29:
If configure is still failing and your system has no /usr/bin/cc link try this:
./configure --CC=/usr/bin/gcc

StefanSchippers, 2024-06-23:
No follow up, closing.

closed 2024-06-23 by StefanSchippers