StefanSchippers/xschem #213: macOS: main window hangs immediately

martinjankoehler, 2024-06-03:
Hi @StefanSchippers 

built xschem on MacBook, the command line only use works,
but when the GUI should be started, the program just hangs.

Is this a known issue?

Data
- macOS 14.5 "Sonoma"
- XQuartz 2.8.5 (other apps like xterm work)
- cairo 1.18.0
- jpeg 9f
- TCL-TK 8.6.14

Directly after calling `xschem` (~/.xschem/xschemrc only contains comments) the process is running with 100% CPU:
[image]

[image]

Best regards,
Martin

StefanSchippers, 2024-06-09:
Ensure you build xschem with a custom installation of the tcl/tk libraries that draw using the X11 emulation (XQuartz on Mac), not the standard tcl/tk libraries that draw on Mac native display (Quartz IIRC)

martinjankoehler, 2024-06-10:
Thanks @StefanSchippers, I confirm this works now with the custom TCL/TK build! 👍 

The only thing I want to note is that the path to the shared library `libtk8.6.dylib` (persisted in the `xschem` binary) was messed up,
so `otool -vL $(which xschem)` (= quasi `ldd`) gave me:
```
/usr/local/bin/xschem:
	/opt/homebrew/opt/jpeg/lib/libjpeg.9.dylib (compatibility version 16.0.0, current version 16.0.0)
	time stamp 2 Thu Jan  1 01:00:02 1970
	/opt/homebrew/opt/readline/lib/libreadline.8.dylib (compatibility version 8.2.0, current version 8.2.0)
	time stamp 2 Thu Jan  1 01:00:02 1970
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1345.120.2)
	time stamp 2 Thu Jan  1 01:00:02 1970
	/opt/X11/lib/libcairo.2.dylib (compatibility version 1.0.0, current version 1.17.6)
	time stamp 2 Thu Jan  1 01:00:02 1970
	/opt/X11/lib/libX11.6.dylib (compatibility version 11.0.0, current version 11.0.0)
	time stamp 2 Thu Jan  1 01:00:02 1970
	/opt/X11/lib/libXrender.1.dylib (compatibility version 5.0.0, current version 5.0.0)
	time stamp 2 Thu Jan  1 01:00:02 1970
	/opt/X11/lib/libxcb.1.dylib (compatibility version 3.0.0, current version 3.0.0)
	time stamp 2 Thu Jan  1 01:00:02 1970
	/opt/X11/lib/libxcb-render.0.dylib (compatibility version 1.0.0, current version 1.0.0)
	time stamp 2 Thu Jan  1 01:00:02 1970
	/opt/X11/lib/libX11-xcb.1.dylib (compatibility version 2.0.0, current version 2.0.0)
	time stamp 2 Thu Jan  1 01:00:02 1970
	/opt/X11/lib/libXpm.4.dylib (compatibility version 16.0.0, current version 16.0.0)
	time stamp 2 Thu Jan  1 01:00:02 1970
	/usr/local/opt/tcl-tk/lib/libtcl8.6.dylib (compatibility version 8.6.0, current version 8.6.10)
	time stamp 2 Thu Jan  1 01:00:02 1970
	/usr/local/opt/tcl-tk/lib:/opt/X11/lib/libtk8.6.dylib (compatibility version 8.6.0, current version 8.6.10)
	time stamp 2 Thu Jan  1 01:00:02 1970
```

Not sure where in the build process this happens, but this line is wrong:
```
	/usr/local/opt/tcl-tk/lib:/opt/X11/lib/libtk8.6.dylib (compatibility version 8.6.0, current version 8.6.10)
```

The way to work around it in post is:
```
sudo install_name_tool -change "/usr/local/opt/tcl-tk/lib:/opt/X11/lib/libtk8.6.dylib" /usr/local/opt/tcl-tk/lib/libtk8.6.dylib $(which xschem)
```

Not sure, but perhaps this could be fixed somewhere in the build scripts?

StefanSchippers, 2024-06-10:
I had this issue with another user on MacOS if I remember correctly. Please see this thread:
https://github.com/StefanSchippers/xschem/issues/24
and go down close to the end:

> Will add a note about the change in Tk Makefile as you suggested...
> The issue is caused by a bug in the tk build script, the wrong path is in wish8.6. In the Makefile of tk8.6.10 the following needs to be changed:
> 
> LIB_RUNTIME_DIR = $(libdir):/opt/X11/lib
> 
> to
> 
> LIB_RUNTIME_DIR = $(libdir)

closed 2024-06-10 by martinjankoehler