StefanSchippers/xschem #215: Manual loading of raw files

georgtree, 2024-06-06:
Hello!
Is it possible to switch off autoloading of raw files when they are specified in the graphs explicitly?
Now if I load one raw file manually, the rest of them are loaded too.
I want to fully control which files are loaded and in what sequence, to work with them later.
Thank you in advance!

StefanSchippers, 2024-06-07:
I will improve that, this will take some time. Previously xschem allowed only one raw file to be loaded. Now multiple files can be loaded, but the load/unload mechanism is still primitive.

StefanSchippers, 2024-06-23:
Graphs now have an auto load checkbutton that loads waves automatically or not. Please test.
[image]

georgtree, 2024-06-23:
Hello, just tested, all works, thank you very much, this option allows more flexible annotation
of schematics :)

georgtree, 2024-07-08:
Hello!
Is it possible to manually load multiple raw files without auto load tick set?
Now I can load multiple files only with this tick, if I load multiplefiles with xschem raw read, it remove previous raw file from the memory.
Thank you.

closed 2024-07-08 by georgtree