StefanSchippers/xschem #216: tcleval called with two arguments instead of 1 when on windows.

Kreijstal, 2024-06-09:
```
# gcc -c -I /ucrt64/include/tk8.6/tk-private/win -I /ucrt64/include/tk8.6/tk-private/generic draw.c
draw.c: In function 'print_image':
draw.c:157:8: error: too many arguments to function 'tcleval'
  157 |   else tcleval("convert_to_png {%s} plot.png", psfile);
      |        ^~~~~~~
In file included from draw.c:23:
xschem.h:1557:20: note: declared here
 1557 | extern const char *tcleval(const char str[]);
      |                    ^~~~~~~
```
where is tcleval with two parameters defined/declared?

StefanSchippers, 2024-06-09:
Don't know why this was not flagged by the windows maintainer.
Thank you for pointing this out.
Fixed

Kreijstal, 2024-06-09:
because you avoid this path if you import cairo :)

closed 2024-06-09 by Kreijstal

StefanSchippers, 2024-06-09:
Thank you for the tip!