StefanSchippers/xschem #218: instance value substitution in `spice_sym_def`

rroohhh, 2024-06-12:
Hi,
It seems to me that currently using instance values (like in `format`) is not supported in `spice_sym_def`. 
I would like to be able to do something like this:
```
type=subcircuit
format="@name @@A @@B @symname"
spice_sym_def=".include @symname_@corner.txt"
template="name=x1 corner=0"
```
To change the `spice_sym_def` depending on a `corner` property set on a instance. 

If implementing this is not desired, is there some other way to achieve this?

StefanSchippers, 2024-06-12:
replace spice_sym_def with device_model and change symbol type from subcircuit to primitive.

```
type=primitive
format=...
template=...
device_model=".include @symname\\_@corner\\.txt"
```

note the double backslashes, they represent a single escape inside the quoted string.
The first escape is needed to tell xschem the first parameter is @symname and not @symname_, the second escape tells the second parameter is @corner and not @corner.txt

inside spice_sym_def no @ substitution is done. I think I will add this feature in a next commit.

StefanSchippers, 2024-06-20:
I made another commit to process @symref (return full path of symbol, like /a/b/c/inv.sym) as well as @symname return only name of symbol with no .sym extension) in spice_sym_def attributes.
Please test if you have a chance to do it.

closed 2024-07-08 by StefanSchippers