StefanSchippers/xschem #220: Ngspice issue

mcmurphy16, 2024-06-22:
I have successfully installed xschem, and tried to run an already given circuit in the library. I'm able to generate the netlist but I'm not able to simulate. I'm attaching a screenshot, please help.

[image]

georgtree, 2024-06-22:
I suspect that the issue is due to missing ngspice executable in PATH, could you run ngspice from terminal using the same command?

mcmurphy16, 2024-06-22:
Thanks for your response @georgtree. No, I don't think that ngspice is installed. Do I need to install it in some specific directory? Also can you share any resources to install ngspice on Ubuntu?

georgtree, 2024-06-22:
There is no latest ngspice version distribution for Ubuntu as far as I know (on the ngspice official website). You can compile it from scratch,
but there are available packages in some Ubuntu repositories: https://packages.ubuntu.com/search?keywords=ngspice
(google how to install from them)

StefanSchippers, 2024-06-23:
This xshem manual page exlains how to install ngspice from the git sources:
https://xschem.sourceforge.io/stefan/xschem_man/tutorial_xschem_sky130.html
```
## clone the source repository into a local ngspice_git directory
git clone https://git.code.sf.net/p/ngspice/ngspice ngspice_git
cd ngspice_git
mkdir release
./autogen.sh
cd release
## by default if no --prefix is provided ngspice will install under /usr/local/{bin,share,man,lib}
## you can add a --prefix=/home/username to install into your home directory.
../configure --with-x --enable-xspice --disable-debug --enable-cider --with-readline=yes --enable-openmp --enable-osdi
## build the program
make
## install the program and needed files.
sudo make install
```

closed 2024-08-06 by StefanSchippers