StefanSchippers/xschem #221: .subckt in format section

georgtree, 2024-06-23:
Hello!
I tried to define custom subcircuit directly in format section like this:
`type=source
format="X@name sub_@name @@in @@out
.subckt sub_@name in out
@name out 0 V=@FUNC 
.ends sub_@name"
template="name=B1 FUNC=\"pow(V(in),2)\""
`
And it expands perfectly fine to this:
`XB1 sub_B1 net1 net2
.subckt sub_B1 in out
B1 out 0 V=pow(V(in),2)
.ends sub_B1`

So my question is - what could go wrong in such way to define custom subcircuit?
It can be an alternative way to pass complex parameters inside .subckt that could not be passed inside ngspice with .param (it doesn't support parameter type except numerical one)

StefanSchippers, 2024-06-23:
I think this is another clever way to create subcircuits from instances. Will add a paragraph to [this manual page](https://xschem.sourceforge.io/stefan/xschem_man/tutorial_instance_based_implementation.html).

StefanSchippers, 2024-06-24:
The only thing that could cause problems with some simulators is that this approach creates nested subcircuits. 
If you place the above component inside a subcircuit you will get something like this:
```
* expanding   symbol:  comp_ngspice.sym # of pins=3
** sym_path: /home/schippes/xschem-repo/trunk/xschem_library/ngspice/comp_ngspice.sym
** sch_path: /home/schippes/xschem-repo/trunk/xschem_library/ngspice/comp_ngspice.sch
.subckt comp_ngspice PLUS OUT MINUS  OFFSET=0 AMPLITUDE=5 GAIN=100 ROUT=1000 COUT=1p
*.ipin PLUS
*.ipin MINUS
*.opin OUT
B1 IOUT 0 V = {OFFSET + AMPLITUDE/2*(tanh(V(IPLUS,IMINUS)*GAIN*2/AMPLITUDE))}
XR1 OUT IOUT sub_R1 Res={ROUT} m=1

.subckt sub_R1 P M Res=1k
R1 P M {Res}
.ends

C3 OUT 0 {COUT} m=1
V1 IPLUS PLUS 0
V2 IMINUS MINUS 0
.ends
```

in the above I have redefined the res.sym to be a subcircuit using your method:

```
type=resistor
format="X@name @@P @@M sub_@name Res=@value m=@m
.subckt sub_@name P M Res=1k
R1 P M {Res}
.ends
"
template="name=R1 value=1k m=1"
```

 You see the .subckt sub_R1 defined inside the subcircuit of the parent schematic. This seems to work in ngspice, but I don't know if this is allowed in all simulators.

StefanSchippers, 2024-06-24:
The following example:
```
test
V1 A 0 3V
X1 A B sub1 RES=100
X2 B C sub1 RES=1000
R1 C 0 1k
.subckt sub1 N1 N2 RES=1
XR1 N1 N2 sub_R1 R={RES}
.subckt sub_R1 P M R=1
R1 P M {R}
.ends
.ends
.op
.end
```

will be flattened by ngspice (`listing r` comand at ngspice prompt) to:
```
* test
v1 a 0 3v
r.x1.r.xr1.r1 a b   1.000000000000000e+02   
r.x2.r.xr1.r1 b c   1.000000000000000e+03   
r1 c 0 1k
.op
.end
```

so you see the x1 and x2 subcircuits will be flattened to 100 and 1000 ohm resistors respectively, so expansion of nested subcircuits works as expected in ngspice

StefanSchippers, 2024-06-24:
I have verified the above example works fine also in Xyce so it is fair to assume nested subcircuits are OK.

georgtree, 2024-06-24:
Hello, great to hear, this approach solves some issues for me, thank you!

closed 2024-06-24 by georgtree

StefanSchippers, 2024-06-24:
I have added the method description at the end of this man page:
https://xschem.sourceforge.io/stefan/xschem_man/tutorial_use_existing_subckt.html