StefanSchippers/xschem #222: 'ngspice: no such file or directory

abdelrahmanali15, 2024-06-25:
## Error when Pressing Simulation button 
[image]

but in the same terminal I can run the simulation manually with the same command 

[image]

I don't know why does this happen. If you have an insight.

StefanSchippers, 2024-06-25:
In xschem go to Simulation ->Configure simulators and tools and replace the word ngspice with the full path of the executable as it is installed on your system.
[image]

abdelrahmanali15, 2024-06-25:
Working Now thank you for your help

closed 2024-06-25 by abdelrahmanali15