StefanSchippers/xschem #224: Xschem with Sky models won't launch

Rafahkz, 2024-07-03:
Hello!

I'm having some trouble when trying to launch xschem with SkyWater models. In my other computer, when I type "xschem &" it runs, but in the PC I'm using, nothing happens... Maybe it is a simple question, but I don't know what to do. Thanks!

[image]

StefanSchippers, 2024-07-04:
- Do you have other graphical applications working on this machine (xclock)?
- What happens if you do `xschem` without the '&'?
- What is printed if you do `xschem -v` ?

Rafahkz, 2024-07-04:
Hi, Stefan. Thanks for the repply.

1) I'm currently not using other graphical applications in this machine;

2)  If use only "xschem", the same thing happens, but the application won't launch;
[image]

3) With this command, the CLI displays
[image]

StefanSchippers, 2024-07-04:
Ok you have a very old xschem package instaled with apt.
Remove the installed xschem:

- sudo apt purge xschem

Then follow this tutorial to install xschem: https://xschem.sourceforge.io/stefan/xschem_man/install_xschem.html

Rafahkz, 2024-07-05:
Hey, Stefan!

In my other computer, it works fine! Even with the older version
[image]

Now I'm on vacation, but as soon as I come back to that first computer, I will try the tutorial you mentioned. Thanks for your answers!

closed 2024-08-06 by StefanSchippers